u-root/u-root #643: Load modules and dependencies

pwFoo, 2018-02-18:
Loading simple modules like ext4 works fine, but during my tests with runc I failed with iptables / nat / bridge needed modules. 

1. I have to solve dependencies manually? Shouldn't modprobe solve / load deps?
1. At the moment modinfo command is missing. Is it planned to add?
1. How to detect hardware and load the modules? I know how to do it with bash / shell, but don't know how to do it with GO or rush :(

Some details about my docker @ runc test in #508

bensallen, 2018-02-18:
1. Yes it should, https://github.com/u-root/u-root/blob/master/pkg/kmodule/kmodule_linux.go#L168
2. It would be useful, but I'm not aware of anyone working on it currently.
3. We need something that parses depmod's module.dep or more ideally module.dep.bin in Go. Then we can start with something basic that iterates over all the node's PCI vendor and device IDs.

pwFoo, 2018-03-07:
Looks like a flag combo problem #326

pwFoo, 2018-04-13:
Modinfo would bei nice, butcI would really need something like this bash script to load modules (with dependencies or run it twice to "solve" deps)
```
grep -h MODALIAS /sys/bus/*/devices/*/uevent | cut -d= -f2 | xargs /sbin/modprobe -abq 2> /dev/null
```

pwFoo, 2018-04-29:
Looks like there is a problem with module loading described in #722.

pwFoo, 2018-05-02:
Anyone tested to load some modules and have problems / errors as described in #722?

pwFoo, 2018-08-26:
some dependencies have to be fixed manually...

closed 2018-08-26 by pwFoo

ZiqianXu, 2024-06-28:
Found this issue when exploring golang equivalent of `modinfo`, is there any plan to support it?
Thanks for your hard work on providing the nice u-boot lib!