u-root/u-root #649: bb mode: only u-root vendor directory is used

hugelgupf, 2018-02-23:
Let's say you are building u-root with Nash, or github.com/hugelgupf/foo.

Now, foo includes a dependency on github.com/hugelgupf/bar, which is vendored in foo, but not in u-root.

When you build u-root in bb mode with foo, only vendored dependencies in u-root are used; the foo ones are ignored. You also get an import error if you don't happen to have bar in your GOPATH.

There are two issues here: 
 * Dependency in external repo that u-root doesn't vendor.
 * Conflicting dependencies. E.g. u-root vendors netlink @ abcdef, but foo vendors netlink @ defghb.

Here's a potential, kind of complicated proposal: 
 * bb mode builds in $TMPDIR/src/bb rather than $GOPATH/src/github.com/u-root/u-root/bbsh
 * bb mode then uses GOPATH=$TMPDIR
 * bb mode builds a synthesized vendor directory in this env, and rewrites all non-standard imports relative to that.
 * conflicting vendored dependencies are vendored twice with different names (bar1, bar2) and included as bar1 and bar2 in the commands that use them.

cc @rjoleary

ryukinix, 2018-02-23:
Yes, this is really a problem. I tried before embedd nash with u-root and I can confirm, this happens.

hugelgupf, 2019-09-23:
This was solved at some point.

closed 2019-09-23 by hugelgupf