u-root/u-root #650: u-root command broken again for -files /bin/tar

rminnich, 2018-02-26:
If I want to add a standard command, e.g. /bin/tar

rminnich@uroot:~/go/src/github.com/u-root/u-root$ go run u-root.go  -build=bb -files /bin/tar
2018/02/26 09:50:34 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/home/rminnich/gohead GOPATH=/home/rminnich/go CGO_ENABLED=0
2018/02/26 09:50:34 Filename is /tmp/initramfs.linux_amd64.cpio
2018/02/26 09:50:44 couldn't add "/bin/tar" to archive: archive path "/bin/tar" must not be absolute (host source "/bin/tar")
exit status 1

note: this used to work. The error is incorrectly passing an abs path to cpio. I'll try to see what got broken. We need to add this command to the circleci as a test. I'm tired of fixing it :-)

hugelgupf, 2018-02-26:
I think somehow in the last couple PRs, the : syntax was made mandatory when it shouldn't have been.

I.e. this works: `-files /bin/tar:bin/tar` but is super annoying.

insomniacslk, 2018-02-26:
I believe I have introduced this issue, I can fix it to support both old and current syntaxes

insomniacslk, 2018-02-26:
following up on https://github.com/u-root/u-root/issues/651

hugelgupf, 2018-02-28:
see https://github.com/u-root/u-root/issues/651#issuecomment-368598475 for what you should be able to do.

closed 2018-03-19 by insomniacslk