u-root/u-root #651: u-root.go: The -files command is now very inconvenient

rminnich, 2018-02-26:
the way the command used to work: 
-files glob[,more globs]
the way it works now
-files file -files file -files file

This only seems more convenient; in actual use I'm finding it difficult.
First, I can't pass a glob to u-root with -files, which is very annoying.

Second, consider a program which calls u-root which also wants to specify a set of files. The user might do this:
program -files x -files y -files z

Now the user program has to additionally use the u-root multiflag, because this won't work with flags. Or it can do what the original u-root program do, but geez, why not skip this mess and just have u-root work the way it used to?

I think we should go back to the original setup, where -flags indicated a list of globs that could be broken apart on some reasonable identifier, e.g. comma.

I'm willing to be argued out of this, but for moment, I'm not finding it usable.

hugelgupf, 2018-02-26:
Yeah, we should improve this again. I think this regressed a bit.

We should not require the : syntax, and we should not require multiple -files flags, we should just offer them additionally.

hugelgupf, 2018-02-26:
I.e. you should be able to do this:

`-files ./cmds/* -files /bin/tar:bin/bar -files "./glob1/* ./glob2/*"`

insomniacslk, 2018-03-01:
The regression has been addressed in https://github.com/u-root/u-root/pull/660 .

Regarding the globbing, I am not sure how to add (and if it's possible at all) in Go a flag that behaves like `-files /some/path/*`, i.e. with multiple separate arguments, since only the first argument would be considered for the `-flags` switch.

It is surely possible instead to leave the globbing to `u-root`, with a syntax like `-files "/somepath/*"` (i.e. the non-expanded string would be passed to the program, and the expansion would happen internally). I am not a fan of the latter syntax but I don't have strong opinions, so if you think this is better I will go with it

closed 2018-03-25 by hugelgupf