u-root/u-root #652: u-root: dhclient does not receive response packets.

dancrossnyc, 2018-02-26:
I can see that they're being generated and sent by the server, but they are not being received by u-root's dhclient:

% dhclient
[   22.768883] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[   22.768883] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
2018/02/26 19:17:05 listen udp6 [fe80::1eb7:2cff:feee:534c%eth0]:546: socket: address family not supported by protocol
2018/02/26 19:17:20 didn't get a packet
2 dhclient attempts were sent%

hugelgupf, 2018-02-26:
I need to switch dhclient back to AF_PACKET. Right now it's a UDP-dgram socket. So e.g. iptables can block the response packet if you have them set up.

hugelgupf, 2018-02-26:
What's your setup here? u-root in a VM, I'm assuming? If you tcpdump in the VM, do you see the packet?

dancrossnyc, 2018-02-26:
My setup: u-root running on hardware.

% uname -a
Linux (none) 4.12.7 #2 SMP PREEMPT Thu Feb 22 17:32:29 EST 2018 x86_64 (none)
% 

The destination machine is also hardware and running OpenBSD 6.2.

rminnich, 2019-09-18:
I'm closing this b/c on hardware I'm not seeing this error and I suspect it has been fixed.

closed 2019-09-18 by rminnich