u-root/u-root #682: A u-root error package.

hugelgupf, 2018-03-14:
Some reading regarding errors: https://commandcenter.blogspot.com/2017/12/error-handling-in-upspin.html

Some of our commands and packages currently do not spit out useful information on error or success.

Printing 5 pages of debug information isn't useful when it's not immediately discernible what the error was -- and for users who aren't developers is even more frustrating because they have to go read the code.

I'm not exactly sure what a u-root systems error package would look like, but I think we want one. Something that annotates errors with some metadata, like which package the error comes from. I think that means that at each layer we'll want to nest errors that are returned in another error and return that. Something like...

pkg/kexec
```go
func Kexec(...) error {
  return uroot.E("kexec", fmt.Errorf(...))
}
```

pkg/boot
```go
func Boot(config) error {
  ...
  if err := Kexec(...); err != nil {
    return uroot.E("boot", err)
  }
}
```

u-root errors should have some utility functions like...

```
func SyscallErrno(err error) syscall.Errno {
  switch e := err.(type) {
  case *uroot.Err:
    // Look at nested error.
    return SyscallErrno(e.Error)

  case syscall.Errno:
    return e

  case *os.PathError:
    ...

  default:
    return 0
  }
}
``` 

Thoughts?

hugelgupf, 2019-02-08:
cc @rjoleary ?

rjoleary, 2019-02-20:
There's a draft for error handling in Go 2: https://go.googlesource.com/proposal/+/master/design/go2draft.md

hugelgupf, 2019-09-23:
Go1.13 errors will likely solve this for us when we start adopting them for real and move to Go 1.13.

closed 2019-09-23 by hugelgupf