u-root/u-root #710: kernel panic "No working init found" with latest u-root?

pwFoo, 2018-04-08:
Hi,

I tested u-root earlier in the year and it works fine and booting up to a shell.
Today I tested it again, but get an kernel panic
```
Failed to execute /init (error -2)
Kernel panic - not syncing: No working init found. [...]
```

Tested with bb and source mode. Build commands like that
```
u-root -build=bb ./cmds/*
u-root -build=source ./cmds/*
u-root -build=bb 
```

Any idea or known bug?

hugelgupf, 2018-04-08:
Might be the thing I just merged. Let me spin up qemu again.

hugelgupf, 2018-04-08:
Can you share the initramfs that it created? I just did a 

./u-root -build=bb

at HEAD and ran it in qemu like this

qemu-system-x86_64 -kernel ... -initrd ... -nographic

and it worked fine. Can you repro in qemu?

pwFoo, 2018-04-08:
@hugelgupf 
Strange ... I have tested it many times in different variants (booted with qemu to test), but it can't reproduce after a reboot of my notebook ...

Because I can't reproduce the problem I close the issue.

Thanks!

closed 2018-04-08 by pwFoo