u-root/u-root #712: Extend modprobe command with -a and -va

pwFoo, 2018-04-08:
Any plans to add `-a` flag and (maybe a workaround for) `-va` flag combo to modprobe?
Played with u-root and docker, but it tries to load modules with `modprobe -va` during start.
```
WARN[2018-03-04T22:53:16.815596551Z] Running modprobe bridge br_netfilter failed with message: flag provided but not defined: -va
Usage of modprobe [-n] modulename [parameters...]:
  -n    Try run
, error: exit status 2 
```

Some changes done to the mounts (cgroup, proc) some month ago. So maybe `modprobe -va` could be the last step to get docker up and running with u-root?

#508, #326

hugelgupf, 2018-04-08:
As a workaround, this is dumb, but we could add a "va" flag to modprobe that just turns on both.

pwFoo, 2018-04-08:
Hi @hugelgupf, would be fine for me. If You wouldn't commit such a workaround it would also be fine to get the needed changes to get it work. I think changes should be in https://github.com/u-root/u-root/blob/master/cmds/modprobe/modprobe_linux.go?

hugelgupf, 2018-04-11:
I'd totally commit such a workaround. Subhav, this one should be a good starter?

subhav, 2018-04-12:
Thanks, I'll look into it.

Re flags: Long-term, have we considered the pflag library in commands? https://github.com/spf13/pflag appears to be the current source-of-truth, and doesn't have any external dependencies.

hugelgupf, 2018-04-13:
@subhav, see also the discussion in #326 for a previous discussion on pflags. If it's a small dependency, we might just as well do it?

pwFoo, 2018-04-15:
I'm new with golang but tried to add `-va` flag quick and dirty. Added before `modName := [...]`
```
        if *loadAll {
                modOptions := ""
                for _, modName := range os.Args[2:] {
                        if err := kmodule.ProbeOptions(modName, modOptions, kmodule.ProbeOpts{DryRun: *dryRun}); err != nil {
                                log.Fatalf("modprobe: Could not load module %q: %v", modName, err)
                        }
                }
                os.Exit(0)
        }
```

After booted up no modules are loaded.
```
% lsmod
Module                  Size  Used by
% 
```

Tried to start docker
```
% dockerd
WARN[2018-04-15T06:56:20.634068824Z] could not change group /var/run/docker.sock to docker: error parsing groups for docker: open /etc/group: no such file or directory 
INFO[2018-04-15T06:56:20.642329554Z] libcontainerd: new containerd process, pid: 113 
WARN[0000] containerd: low RLIMIT_NOFILE changing to max  current=1024 max=4096
WARN[2018-04-15T06:56:20.663289666Z] failed to rename /var/lib/docker/tmp for background deletion: rename /var/lib/docker/tmp /var/lib/docker/tmp-old: no such file or directory. Deleting synchronously 
INFO[2018-04-15T06:56:20.729612880Z] Graph migration to content-addressability took 0.00 seconds 
WARN[2018-04-15T06:56:20.731294337Z] Unable to find blkio cgroup in mounts        
WARN[2018-04-15T06:56:20.732510289Z] Unable to find cpuset cgroup in mounts       
WARN[2018-04-15T06:56:20.733747114Z] mountpoint for pids not found                
INFO[2018-04-15T06:56:20.735229116Z] Loading containers: start.                   
[    7.256528] nf_nat: Unknown symbol nf_ct_ext_add (err 0)
[    7.257716] nf_nat: Unknown symbol nf_ct_invert_tuplepr (err 0)
[    7.258875] nf_nat: Unknown symbol nf_ct_seqadj_set (err 0)
[    7.260464] nf_nat: Unknown symbol nf_ct_helper_expectfn_unregister (err 0)
[    7.261995] nf_nat: Unknown symbol nfnetlink_parse_nat_setup_hook (err 0)
[    7.263374] nf_nat: Unknown symbol nf_ct_extend_register (err 0)
[    7.264549] nf_nat: Unknown symbol nf_ct_helper_expectfn_register (err 0)
[    7.266006] nf_nat: Unknown symbol nf_ct_extend_unregister (err 0)
[    7.267308] nf_nat: Unknown symbol nf_ct_l3proto_module_put (err 0)
[    7.268782] nf_nat: Unknown symbol nf_conntrack_tuple_taken (err 0)
[    7.270196] nf_nat: Unknown symbol nf_conntrack_htable_size (err 0)
[    7.271592] nf_nat: Unknown symbol nf_conntrack_alter_reply (err 0)
[    7.272771] nf_nat: Unknown symbol nf_ct_free_hashtable (err 0)
[    7.274158] nf_nat: Unknown symbol nf_ct_iterate_destroy (err 0)
[    7.275539] nf_nat: Unknown symbol crc32c (err 0)
[    7.276404] nf_nat: Unknown symbol nf_ct_l3proto_try_module_get (err 0)
[    7.277933] nf_nat: Unknown symbol nf_ct_alloc_hashtable (err 0)
Error starting daemon: Error initializing network controller: error obtaining controller instance: failed to create NAT chain: Iptables not found
wait: exit status 1
```

Some modules are loaded by `modprobe -va` during `dockerd`.
```
% lsmod
Module                  Size  Used by
xt_conntrack           16384  0
nf_nat                 32768  0
nf_conntrack          135168  2 xt_conntrack,nf_nat
libcrc32c              16384  2 nf_nat,nf_conntrack
overlay                73728  0
```

Any idea about the unknown symbol messages?
```
[    7.256528] nf_nat: Unknown symbol nf_ct_ext_add (err 0)
[    7.257716] nf_nat: Unknown symbol nf_ct_invert_tuplepr (err 0)
[    7.258875] nf_nat: Unknown symbol nf_ct_seqadj_set (err 0)
[    7.260464] nf_nat: Unknown symbol nf_ct_helper_expectfn_unregister (err 0)
[    7.261995] nf_nat: Unknown symbol nfnetlink_parse_nat_setup_hook (err 0)
[    7.263374] nf_nat: Unknown symbol nf_ct_extend_register (err 0)
[    7.264549] nf_nat: Unknown symbol nf_ct_helper_expectfn_register (err 0)
[    7.266006] nf_nat: Unknown symbol nf_ct_extend_unregister (err 0)
[    7.267308] nf_nat: Unknown symbol nf_ct_l3proto_module_put (err 0)
[    7.268782] nf_nat: Unknown symbol nf_conntrack_tuple_taken (err 0)
[    7.270196] nf_nat: Unknown symbol nf_conntrack_htable_size (err 0)
[    7.271592] nf_nat: Unknown symbol nf_conntrack_alter_reply (err 0)
[    7.272771] nf_nat: Unknown symbol nf_ct_free_hashtable (err 0)
[    7.274158] nf_nat: Unknown symbol nf_ct_iterate_destroy (err 0)
[    7.275539] nf_nat: Unknown symbol crc32c (err 0)
[    7.276404] nf_nat: Unknown symbol nf_ct_l3proto_try_module_get (err 0)
[    7.277933] nf_nat: Unknown symbol nf_ct_alloc_hashtable (err 0)
```
As you can see above nf_nat is loaded.

Additional mounts should be added to boot / init (#508) 
```
WARN[2018-04-15T06:56:20.731294337Z] Unable to find blkio cgroup in mounts        
WARN[2018-04-15T06:56:20.732510289Z] Unable to find cpuset cgroup in mounts       
WARN[2018-04-15T06:56:20.733747114Z] mountpoint for pids not found      
```

subhav, 2018-05-03:
In any case, let's commit -va the way you did, since there don't seem to be any problems actually loading the modules.

pwFoo, 2018-05-03:
@subhav @hugelgupf 
`modprobe -a` and `modprobe -va` seems to work. Thanks!

Other problems with module loading have own issues #722 / #643

closed 2018-05-03 by hugelgupf