u-root/u-root #732: gVisor + u-root distro

hugelgupf, 2018-05-02:
One of those ideas for Go enthusiasts: an "all Go secure container runtime Linux distro". Include kubelet, docker, gVisor, u-root. Go userspace, Go kernel, and Linux ends up being "just" the hypervisor. Basically a "kubelet distro".

rjoleary, 2018-05-23:
Maybe it would be something interesting to see in http://www.qemu-advent-calendar.org/2016/ ? It looks like they have a new calendar every two years.

pwFoo, 2018-07-24:
Any progress with gVisor (runsc / docker) running with u-root? Just shell with gVisor / run(s)c or docker would be nice!

hugelgupf, 2019-02-08:
I'm going to close this as infeasible for now, since gVisor is only buildable with bazel. (Someone has a hacked up non-bazel version, but it's likely to break.)

closed 2019-02-08 by hugelgupf