u-root/u-root #742: kernel panic with latest u-root

pwFoo, 2018-05-10:
Any idea why I get a kernel panic with latest u-root in qemu and with dell laptop hw:
```
[    1.016055] Failed to execute /init (error -2)
[    1.017135] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.
[    1.019923] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 4.16.7-1.el7.elrepo.x86_64 #1
[    1.021667] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.10.2-0-g5f4c7b1-prebuilt.qemu-project.org 04/01/2014
[    1.024494] Call Trace:
[    1.025075]  dump_stack+0x63/0x88
[    1.025853]  ? rest_init+0xb0/0xb0
[    1.026651]  panic+0xe8/0x258
[    1.027345]  ? putname+0x53/0x60
[    1.028095]  ? rest_init+0xb0/0xb0
[    1.028886]  kernel_init+0xf9/0x110
[    1.029697]  ret_from_fork+0x35/0x40
[    1.030670] Kernel Offset: disabled
[    1.031598] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.
qemu-system-x86_64: terminating on signal 15 from pid 17095 (<unknown process>)
```

Also tested earlier versions with same result:
- https://github.com/u-root/u-root/tree/38593319cfa559f058fa9d97c17e8ff8d45dca1a
- https://github.com/u-root/u-root/tree/d997cd8fcefe97ce064ac9759afb2269d3632e8f

pwFoo, 2018-05-10:
a u-root docker image (u-root bin + repo) 6 days old works fine.

pwFoo, 2018-05-10:
Strange... looks host dependent. Tested with another host and it runs without kernel panic.
So I'll close the issue for now.

closed 2018-05-10 by pwFoo

hugelgupf, 2018-05-11:
That's weird.

Did you include the init program in your initramfs? Up until recently, init was automatically included in bb and source mode. You now have to specify it:
```
./u-root -build=bb ./cmds/ls ./cmds/ip ./cmds/init
```

pwFoo, 2018-05-11:
Thanks, maybe part of my problem! But host /  os dependent `cmds/*` it fails too.

But good to know `cmds/init` should be added manually! Thanks.

Instead of multiple `./cmds/ls` short `cmds form like `cmds/{init,IP,...}` or `cmds/*` should still work?