u-root/u-root #748: Able to choose default shell from cmd line when building

hugelgupf, 2018-05-23:
let's give users the option.

hugelgupf, 2018-06-06:
right now, our init tries to exec "sh" and then "rush". https://github.com/u-root/u-root/blob/master/cmds/init/init.go#L37 

you should be able to choose a default shell from the command line. It's harder to actually change the init code dynamically, so let's just settle on running "/bbin/defaultsh" or "/buildbin/defaultsh" in init permanently and adding a symlink dynamically.

i.e.

u-root --default-shell=elvish github.com/u-root/elvish

means the initramfs has the following files in /bbin:
... bb [binary]
... elvish -> bb [symlink]
... defaultsh -> elvish [symlink]

and in init we try to run "defaultsh" first and fall back to "sh" and "rush".

hugelgupf, 2018-06-07:
talking to subhav, realized that doesn't actually 100% work, as "defaultsh" will be the arg[0] and it won't recognize elvish in bb mode. hm. Maybe that means a special thing to make bb also recognize defaultsh as arg[0] to go to your chosen default shell?

rjoleary, 2018-07-28:
The code changes might be similar to https://github.com/u-root/u-root/pull/810

PR #856 (2018-08-06): u-root: -defaultsh for default shell to start after init.
Fixes #748.

E.g.:

  ./u-root -defaultsh=elvish github.com/u-root/elvish \
                      github.com/u-root/u-root/cmds/*

  ./u-root -defaultsh=/bin/bash -files=/bin/bash

This will create a symlink /bin/defaultsh pointing to the appropriate
shell.

In source mode, init will resolve one level of the symlink before
executing the command in order to achieve the correct argv[0]. E.g. if
the symlink chain is:

  /buildbin/defaultsh -> /buildbin/rush -> /buildbin/installcommand

Then to compile rush, argv[0] has to have the suffix " [... cut]
commit f098aead (2018-08-06): u-root: -defaultsh for default shell to start after init.

In the uroot.CreateInitramfs Go API, you now have to specify a
DefaultShell. However, in init, sh and rush are still the fallbacks.

Fixes #748.

E.g.:

  ./u-root -defaultsh=elvish github.com/u-root/elvish \
                      github.com/u-root/u-root/cmds/*

  ./u-root -defaultsh=/bin/bash -files=/bin/bash

This will create a symlink /bin/defaultsh pointing to the appropriate
shell.

In source mode, init will resolve one level of the symlink before
executing the command in order to achieve the correct argv[0]. E.g. if
the symlink  [... cut]
closed 2018-08-06 by hugelgupf