u-root/u-root #761: which needs to be smarter on Linux, it gives incorrect answers

rminnich, 2018-06-01:
on linux, /usr/bin/which uses faccessat to figure out if something can be run, so:
```
rminnich@uroot:~/projects/titan/tock-teensy$ ls -l /tmp/date
-rwx------ 1 root rminnich 104664 May 31 21:36 /tmp/date
rminnich@uroot:~/projects/titan/tock-teensy$ PATH=/tmp:$PATH which date
/bin/date
rminnich@uroot:~/projects/titan/tock-teensy$ sudo su
[sudo] password for rminnich: 
root@uroot:/home/rminnich/projects/titan/tock-teensy# PATH=/tmp:$PATH which date
/tmp/date
```

becase:

> stat("/tmp/date", {st_mode=S_IFREG|0700, st_size=104664, ...}) = 0
> geteuid()                               = 1000
> faccessat(AT_FDCWD, "/tmp/date", X_OK)  = -1 EACCES (Permission denied)
> ...
> stat("/bin/date", {st_mode=S_IFREG|0755, st_size=104664, ...}) = 0
> geteuid()                               = 1000
> faccessat(AT_FDCWD, "/bin/date", X_OK)  = 0

note that our which is not that smart.
```
rminnich@uroot:~/go/src/github.com/u-root/u-root/cmds/which$ PATH=/tmp:$PATH ./which date
/tmp/date
rminnich@uroot:~/go/src/github.com/u-root/u-root/cmds/which$ /tmp/date
bash: /tmp/date: Permission denied
rminnich@uroot:~/go/src/github.com/u-root/u-root/cmds/which$ 
```

oops.

hugelgupf, 2018-06-01:
using faccessat to figure out if something can be run is *just wrong*, and I am not at all surprised that GNU which gets it wrong.

From the man page:

>  The check is done using the calling process's real UID and GID, rather than the effective IDs as is done when actually attempting an operation (e.g., open(2)) on the file.  Similarly, for the root user, the check uses the set of permitted capabilities rather than the set of effective capabilities; and for non-root users, the check uses an empty set of capabilities.

> This  allows  set-user-ID programs and capability-endowed programs to easily determine the invoking user's authority.  In other words, access() does not answer the "can I read/write/execute this file?" question.  It answers a slightly different question: "(assuming I'm a setuid binary) can the user who invoked me read/write/execute this file?", which gives set-user-ID programs the possibility to prevent malicious users from causing  them  to  read files which users shouldn't be able to read.

rminnich, 2018-06-01:
chris just cleared me up, but it's depressing.

closed 2018-06-01 by rminnich

hugelgupf, 2018-06-01:
neither the first order approximation i chose nor faccessat really capture the whole thing (besides the fact that it's racy). faccessat may indeed capture stuff like MS_NOEXEC in the mount/superblock flags, but then uses the wrong UID to check. crapshoot either way. maybe we should go with bug-for-bug compatible anyway.