u-root/u-root #766: Split packages and commands into camps

hugelgupf, 2018-06-14:
u-root is starting to get more and more code, and we intuitively seem to assign it importance. Additionally, there is some stuff here that we are likely to put in production, some stuff that we are likely to use only for debugging, some stuff that doesn't apply to all u-root use cases, and some stuff that is pretty experimental.

We would like some structural guarantees that things that go into production meet a higher standard of quality than other parts of the code base.

Here's the proposal we got to after a heated slack discussion (with some additions from me):

- Create an "x" directory for cmds and packages that we consider experimental or non-essential.
  - E.g. x/pkg/sos, x/cmd/wifi
  - the existing cmds/ and pkg/ will stay around, but only "core" commands will be there.
- Ensure better test coverage for core cmds, everything that isn't moved to x.
  - We can make it some kind of bar like 80%, though test coverage is a somewhat arbitrary measure anyway. The bar should be _meaningful tests_. We can make 80% the benchmark to look for (the minimum effort to put in), but we can deviate from the rule as we like.
  - Gotta get some automatic test coverage reporting in here, like coverity.
  - I would like golint (with a few features turned off) to be turned on for the core things. That means nice documentation, etc.
- Let's say we want to keep "cmds/wifi" in the core, but nobody is willing to give it better test coverage (e.g. because "we have more important things to worry about"). _If we don't feel like better test coverage is worth the effort for a particular cmd, it belongs in x._ When some day someone changes their mind, it can graduate from x.
  - But that means new features have to be up to the higher standard as well.

We should only put in production what's in core. Want to put something in prod? Gotta put in the effort to graduate it from "x".

One other thing -- I have some more things coming for kexec/booting (crazy experiments incoming, hrrm). I think maybe the currently empty github.com/u-root/netboot21 repo needs to become github.com/u-root/boot21 and pkg/kexec, pkg/boot, cmd/diskboot, cmd/pxeboot, etc should move there. It's somewhat specific to LinuxBoot, and that stuff probably doesn't need to be in NiChrome or a BMC distro, for example (maybe cmds/kexec would stay, but the others don't need to be there).

cc @insomniacslk @rminnich @GanShun @dhendrix @zaolin

rjoleary, 2018-06-14:
I like the idea, but I think the ordinary solution of using git's branching is worth considering first.

Once a command matures, cherry-pick it onto the master branch (and nicely preserve history). Our build system will not need to change one bit and we can use the robust branching facilities of git, github and CI. Additionally, this allows for adding new unstable features to existing production-ready commands.

Go-getting the production version remains the same, and switching to develop is straight-forward:

```
cd "$GOPATH/src/github.com/u-root/u-root"
git checkout develop
go install
```

Many go projects use branching. For example, https://github.com/gohugoio/hugo has a master and stable branch.

hugelgupf, 2018-06-14:
I don't know that I'm a fan, just because of discoverability. I want people to be able to click through the tree on GitHub and see everything. People usually don't think to look at other branches, just from a GitHub UI POV.

rjoleary, 2018-06-14:
> I don't know that I'm a fan, just because of discoverability. I want people to be able to click through the tree on GitHub and see everything. People usually don't think to look at other branches, just from a GitHub UI POV.

It's possible to change github's default branch under Settings -> branches.

Alternatively, "master" can be used for development (everyone will see all the commands by default) and "stable" is used for production.

rminnich, 2018-06-14:
I'm ok either way. 

The one attractive piece of this idea is that we can take it a bit further. For example, we might have this:
cmds/core/...
pkg/core/...
pkg/x/...
cmds/x/...
cmds/NiChrome/...
pkg/NiChome/...

so we'd have more classes of things. This is hard to do with branches.

rminnich, 2018-06-14:
also, note, I did flip Chris's paths above, I'm fine with what we have:
cmds/...
pkg/...
x/cmds/...
x/pkg/...

whatever is easiest, and I bet Chris's idea is easiest. 

For a first pass, we can toss the non-covered crap into x and see if it seems ok. 

And thank goodness we avoid overlays ...

rjoleary, 2018-06-14:
I concede either way will work. Go with whichever most people are comfortable with :)

rjoleary, 2018-07-13:
Fixed by #804

closed 2018-07-13 by rjoleary