u-root/u-root #787: Kernel requirements?

prologic, 2018-06-29:
I noticed booting up an initramfs in source mode that the bootup complained of a bunch of missing cgroup paths in `/sys/...`. What are the kernel requirements? Same as for Docker as documented [here](https://github.com/moby/moby/blob/master/contrib/check-config.sh)?

prologic, 2018-06-29:
Can we update the README/docs to reflect this if so?

hugelgupf, 2018-06-29:
We should make those shut up. We automatically try to mount them, but they're not necessary for anything u-root-specific.

hugelgupf, 2018-06-29:
@rminnich has been working on a separate kernel requirement doc.

prologic, 2018-06-30:
So if these are **not** are hard requirement; I've had little luck booting into anything useful. What should be the end result here? Drop into a shell? A getty-style login? What? (*I didn't see any of this; just errors*)

hugelgupf, 2018-06-30:
You should see a shell prompt, %. Is it reproducible in qemu? (Not sure you can share the kernel you're using, though.)

prologic, 2018-06-30:
It is; Let me spend some time building a branch that repros this.

James Mills / prologic

E: prologic@shortcircuit.net.au
W: prologic.shortcircuit.net.au

hugelgupf, 2019-02-08:
We've documented some of the configs we use in tests here https://github.com/u-root/u-root/tree/master/configs.

We should rewrite that to be the u-root configs, but there's something. (to be improved.)

closed 2019-02-08 by hugelgupf