u-root/u-root #805: Migrate from dep to vgo

rjoleary, 2018-07-15:
It appears vgo is the future...

"The vgo proposal has been accepted and will be merged in the Go tree and replace dep at a later date. You will be able to experiment with the vgo workflow from Go 1.11 as it will be included as an experiment in this release." ([source](https://github.com/golang/go/wiki/vgo))

This is not urgent. Wait until the 1.11 release and migration guides before starting anything.

hugelgupf, 2018-07-27:
This is going to change the bb build process in some way, so this is not trivial. We'll have to see.

rjoleary, 2018-08-16:
The releases are now being tagged in the format required by vgo, v1.0.0. When go1.11 is released (sometime in the next month), we can start by migrating Nichrome or fiano first (they both depend on u-root).

Nichrome and fiano have a much more typical build process when compared to u-root. Once they are migrated over, we can start thinking about u-root.

hugelgupf, 2019-05-15:
I was thinking that an initial change to the build process could use `go mod vendor` in all relevant directories (not just u-root if you're getting cmds from outside u-root) and build as we do right now.

hugelgupf, 2019-09-23:
In progress in #1358

closed 2024-02-20 by hugelgupf