u-root/u-root #834: bb: Command flag contamination

rjoleary, 2018-07-24:
Today, I saw a `-soshtml` flag on the `kexec` command while running in bb mode. The flag should not be there.

rjoleary, 2018-07-27:
The `-soshtml` isn't an issue with bb; it's actually in a package imported by `init`. We'll need to fix-up the package.

rjoleary, 2018-07-27:
@hugelgupf Also mentioned he saw a possible contamination with testing flags.

hugelgupf, 2018-07-27:
Contamination comes from pkg/sos, which always declares the soshtml flag.

Contamination also comes from elvish, which imports "testing" in non-test files. testing always declares all the testing flags.

hugelgupf, 2018-07-27:
@TrevorFarrelly can you remove the soshtml flag from pkg/sos and put it in the command it belongs to?

rjoleary, 2018-07-29:
> Contamination comes from pkg/sos, which always declares the soshtml flag.

This is now fixed. Thanks you Jorge and Trevor!

> Contamination also comes from elvish, which imports "testing" in non-test files. testing always declares all the testing flags.

Can we modify our fork of elvish so it doesn't do that?

hugelgupf, 2019-02-08:
hey @rminnich - we still investing in elvish, or ash?

closed 2019-09-23 by hugelgupf