u-root/u-root #842: installcommand: imports runtime: open /go/src/runtime/asm_ppc64x.h: no such file or directory

ddiss, 2018-07-27:
I'm playing around with u-root as a replacement for Dracut in rapido (https://github.com/rapido-linux/rapido)

I currently encounter the following error on boot:
<pre>
/usr/bin/qemu-kvm -smp cpus=2 -m 512 -kernel &ltbzImage&gt -initrd ./initrds/myinitrd -append 'ip=192.168.1.1:::255.255.255.0:rapido1 rd.systemd.unit=emergency rd.shell=1 console=ttyS0 rd.lvm=0 rd.luks=0' -pidfile ./initrds/rapido_vm1.pid -device e1000,netdev=nw1,mac=&ltmac&gt -netdev tap,id=nw1,script=no,downscript=no,ifname=tap0 -nographic
...
package github.com/u-root/u-root/cmds/installcommand
        imports runtime: open /go/src/runtime/asm_ppc64x.h: no such file or directory
2018/07/27 16:25:30 exit status 1
2018/07/27 16:25:30 init: No suitable executable found in [/inito /bbin/uinit /buildbin/uinit /bbin/sh /bbin/rush /buildbin/sh /buildbin/rush]
2018/07/27 16:25:30 init: Waiting for orphaned children
2018/07/27 16:25:30 init: All commands exited
2018/07/27 16:25:30 init: Syncing filesystems
2018/07/27 16:25:30 init: Exiting...
</pre>

The image was generated by current master (569a17126131f2f320dd70ec9151a262e9e25846) on openSUSE Leap 42.3 with go 1.9.7 via:
<pre>
u-root (master)> ./u-root -format=cpio -build=source -o myinitrd ./cmds/{ls,ip,dhclient,cat,init,rush,installcommand}
2018/07/27 18:24:26 Disabling CGO for u-root...
2018/07/27 18:24:26 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib64/go/1.9 GOPATH=/home/ddiss/go:/usr/share/go/1.9/contrib CGO_ENABLED=0
2018/07/27 18:24:26 Filename is initramfs.cpio
2018/07/27 18:24:26 Collecting package files and dependencies...
2018/07/27 18:24:28 Building go toolchain...
2018/07/27 18:24:58 Successfully wrote initramfs.
</pre>

I'm not very familiar with Go, so would appreciate if someone could point me in the right direction here.

ddiss, 2018-07-27:
@hugelgupf Thanks for the feedback.
Please consider changing README.md if that's the case, as it currently states:
> Make sure your Go version is the latest (>=1.9).

hugelgupf, 2018-07-27:
One more question - are you actually on ppc64x? The file's suffix would
indicate so?

ddiss, 2018-07-27:
> No, this is x86_64. The header contains a single FIXED_FRAME #define .

Just to elaborate a little, the runtime/asm_ppc64x.h header file is included with both 1.9 and 1.10 go versions shipped with openSUSE, but I've only tried the procedure on 1.9 so far.

rjoleary, 2018-07-28:
Does it work if you use bb mode instead? To do so, replace `-build=source` with `-build=bb` in u-root's command line.

ddiss, 2018-07-29:
bb mode appears to boot through to rush fine:
<pre>
./u-root -format=cpio -build=bb -o initramfs.cpio ./cmds/{ls,ip,dhclient,cat,init,rush,installcommand}
2018/07/29 17:21:11 Disabling CGO for u-root...
2018/07/29 17:21:11 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib64/go/1.9 GOPATH=/home/ddiss/go:/usr/share/go/1.9/contrib CGO_ENABLED=0
2018/07/29 17:21:11 Filename is initramfs.cpio
2018/07/29 17:21:17 Successfully wrote initramfs.

/usr/bin/qemu-kvm -smp cpus=2 -m 512 -kernel <bzImage> -initrd ./initrds/myinitrd -append 'ip=192.168.1.1:::255.255.255.0:rapido1 rd.systemd.unit=emergency rd.shell=1 console=ttyS0 rd.lvm=0 rd.luks=0' -pidfile ./initrds/rapido_vm1.pid -device e1000,netdev=nw1,mac=<mac> -netdev tap,id=nw1,script=no,downscript=no,ifname=tap0 -nographic
...
2018/07/29 15:32:06 In a break with tradition, you seem to have NO u-root commands: <nil>
2018/07/29 15:32:06 exec: "go": executable file not found in $PATH
% ls
bbin
bin
...
</pre>

rjoleary, 2018-07-29:
I just gave it a try and source mode hangs for me while bb works properly. For now, just use bb mode.

I'll work on including a source mode test into our CI.

ddiss, 2018-07-29:
Thanks for the followup.

ddiss, 2018-08-07:
For the record, I noticed that an asm_ppc64x.h symlink **was** carried in the cpio image, but the target was missing. After adding the target (and a couple of other missing headers), I now get further through the source mode boot:
<pre>
ExtraFiles:[... /usr/lib64/go/1.9/pkg/include/asm_ppc64x.h:lib64/go/1.9/pkg/include/asm_ppc64x.h /usr/lib64/go/1.9/pkg/include/funcdata.h:lib64/go/1.9/pkg/include/funcdata.h /usr/lib64/go/1.9/pkg/include/textflag.h:lib64/go/1.9/pkg/include/textflag.h]

qemu boot:
2018/08/07 17:48:38 Couldn't compile "init": error building go package in "/src/github.com/u-root/u-root/cmds/init": # runtime
/go/src/runtime/asm.s:13: illegal or missing addressing mode for symbol RODATA
/go/src/runtime/asm.s:15: illegal or missing addressing mode for symbol NOPTR
/go/src/runtime/asm.s:16: illegal or missing addressing mode for symbol NOPTR
/go/src/runtime/asm.s:36: illegal or missing addressing mode for symbol NOSPLIT
asm: assembly of /go/src/runtime/asm.s failed
, exit status 2
[    4.577059] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100
</pre>

rjoleary, 2018-08-08:
Thanks for the update David! I didn't realize "asm_ppc64x.h" is a symlink. It's possible the symlink is a recent change to golang and our build system doesn't resolve it correctly.

@rminnich is working on a test for source mode in #854 (WARNING: do not use this test at the moment. It does bad things to Linux and you'll have to restart your machine). He should have the test working properly tomorrow. When it's ready, it'll be interesting to see what the test does with your setup.

rjoleary, 2018-08-18:
@rminnich's testinitramfs script is ready now. You can find it under "scripts/testramfs/testramfs.go". You must build it first, then run as sudo while passing in the CPIO file. It runs u-root inside a container.

hugelgupf, 2018-08-18:
I have a good idea of why this is broken. :)

hugelgupf, 2018-08-18:
#464 is actually the likely solution, once I rebase it and clean it up.

ddiss, 2018-10-18:
FWIW, I still see failures when booting source mode images with go 1.11 on openSUSE. e.g.

./u-root -defaultsh rush -build source -o ./myinit github.com/u-root/u-root/cmds/{init,ls,installcommand} github.com/u-root/u-root/xcmds/rush
2018/10/18 15:24:45 Disabling CGO for u-root...
2018/10/18 15:24:45 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib64/go/1.11 GOPATH=/home/ddiss/go:/usr/share/go/1.11/contrib CGO_ENABLED=0
2018/10/18 15:24:45 Filename is ./myinit
2018/10/18 15:24:45 Collecting package files and dependencies...
2018/10/18 15:24:48 Building go toolchain...
2018/10/18 15:25:24 Successfully wrote initramfs.

qemu boot (with mainline kernel bzimage):
[    0.734924] Run /init as init process
[    1.446595] tsc: Refined TSC clocksource calibration: 2593.988 MHz
[    1.447494] clocksource: tsc: mask: 0xffffffffffffffff max_cycles: 0x25640cb3a67, max_idle_ns: 440795258696 ns
[    1.448877] clocksource: Switched to clocksource tsc
2018/10/18 13:31:21 Couldn't compile "init": error building go package in "/src/github.com/u-root/u-root/cmds/init": # runtime
/go/src/runtime/asm.s:13: illegal or missing addressing mode for symbol RODATA
/go/src/runtime/asm.s:15: illegal or missing addressing mode for symbol NOPTR
/go/src/runtime/asm.s:16: illegal or missing addressing mode for symbol NOPTR
/go/src/runtime/asm.s:39: illegal or missing addressing mode for symbol NOSPLIT
asm: assembly of /go/src/runtime/asm.s failed
, exit status 2

rjoleary, 2018-10-22:
I cannot reproduce. Make sure you run `go build` after checking out the latest commit. Everything before the error message looks normal. Possibly our build system has issues when GOPATH contains multiple entries (I've only ever used one)? Possibly there is some bad interaction with openSUSE?

If possible, can you please include the output of `go env` and attach the generated cpio?

hugelgupf, 2019-02-08:
How's this going?

I feel like we need to setup a circleci test that compiles u-root in some distros like OpenSUSE and Fedora, with their weird symlinking package management setups, which tend to trip up Go.

commit 69d8a7a1 (2018-10-17): initramfs: follow symlinks on host file system.

Should also fix #842.
closed 2019-09-23 by hugelgupf