u-root/u-root #880: Measure KPIs

rjoleary, 2018-08-10:
Metrics tend to get worse over time if they aren't kept in check. It would be nice to have a dashboard which plotted the following Key Performance Indicators (KPIs), one entry per day:

* Size of busybox CPIO in KiB
* Size of compressed busybox CPIO in KiB
* Size of source CPIO in KiB
* Size of compressed source CPIO in KiB
* Size of minimal Linux kernel in KiB
* Boot time for busybox mode in seconds
* Boot time for source mode in seconds

JohnAZoidberg, 2018-09-16:
So basically you want the CI to also do and record these measurements and then have them displayed nicely somewhere?

What is the *minimal* linux kernel that you want to measure?

Should the boot time be with that same minimal kernel using QEMU or only with `testramfs`?

rjoleary, 2021-05-27:
Duplicate of #1809

closed 2021-05-27 by rjoleary