u-root/u-root #899: u-root's sshd has pty issues when it is run in a pty or when it gets demonized.

rminnich, 2018-08-17:
ryan reports:

u-root's sshd has pty issues when it is run in a pty or when it gets demonized.
The way I got it to work is run it directly on /dev/tty1

rminnich, 2018-08-17:
the failure is

hugelgupf :: ssh uroot@fdsv267
Enter passphrase for key '/usr/local/google/home/chrisko/.ssh/id_rsa_nerf_reg1':
PTY allocation request failed on channel 0
yes it is

rjoleary, 2018-08-17:
Here's my report:

This issue arose from using u-root's sshd, compiled statically on an ubuntu-based distro. Ideally, it should be possible to demonize the sshd binary with:

```
( sshd -keys .ssh/authorized_keys -privatekey .ssh/id_rsa -d -port 22 & ) &
```

This successfully demonizes the process, but it unable to create new ptys when receiving connections. I forget the exact wording on the error, but it was something along the lines of not being able to allocate a new pty.

Instead, I was able to use sshd in the background by:

1. Switching to a new virtual tty (for example CTRL-ALT-F1 for /dev/tty1)
2. Logging in
3. `sudo sshd -keys .ssh/authorized_keys -privatekey .ssh/id_rsa -d -port 22`
4. Switching back to whichever terminal I was on before

However, it is quite insecure leaving a tty open like this.

rjoleary, 2018-08-17:
Note: My issue was prior to any of @bluecmd's changes.

bluecmd, 2018-11-04:
I'll close this for now, I'm using SSH quite heavily and I haven't run into this. Feel free to reopen if it's still a problem.

closed 2018-11-04 by bluecmd