u-root/u-root #931: ip address printing broken

bluecmd, 2018-09-09:
On my Laptop:
```
$ ./ip addr
1: lo: <UP,LOOPBACK> mtu 65536 state UNKNOWN
    link/loopback 
    inet <nil> scope host lo
       valid_lft forever preferred_lft forever
    inet6 <nil> scope host 
       valid_lft forever preferred_lft forever
2: enp0s31f6: <UP,BROADCAST,MULTICAST> mtu 1500 state DOWN
    link/ether 54:ee:75:cf:52:9c
3: wlp4s0: <UP,BROADCAST,MULTICAST> mtu 1500 state UP
    link/ether 14:ab:c5:53:fc:55
    inet <nil> brd 10.0.200.255 scope global wlp4s0
       valid_lft 71979sec preferred_lft 71979sec
    inet6 <nil> scope link 
       valid_lft forever preferred_lft forever
```

This is on amd64, 4.17.0 kernel.

insomniacslk, 2018-09-09:
Apparently this[1] should use `addr.IP` instead of `addr.Peer`

[1] https://github.com/u-root/u-root/blob/master/cmds/ip/ops.go#L56

insomniacslk, 2018-09-09:
With the patch below I get the following:

```
$ ./ip a
1: lo: <UP,LOOPBACK> mtu 65536 state UNKNOWN
    link/loopback 
    inet 127.0.0.1 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1 scope host 
       valid_lft forever preferred_lft forever
2: wlp2s0: <UP,BROADCAST,MULTICAST> mtu 1500 state UP
[...]
```

```
diff --git a/cmds/ip/ops.go b/cmds/ip/ops.go
index 9beaeda..c19b0fc 100644
--- a/cmds/ip/ops.go
+++ b/cmds/ip/ops.go
@@ -53,7 +53,7 @@ func showLinkAddresses(w io.Writer, link netlink.Link) error {
                        return fmt.Errorf("Can't figure out IP protocol version")
                }
 
-               fmt.Fprintf(w, "    %s %s", inet, addr.Peer)
+               fmt.Fprintf(w, "    %s %s", inet, addr.IP)
                if addr.Broadcast != nil {
                        fmt.Fprintf(w, " brd %s", addr.Broadcast)
                }
```

bluecmd, 2018-09-09:
Verified, works for me! :+1:

insomniacslk, 2018-09-09:
Fix pushed (https://github.com/u-root/u-root/pull/932), closing

closed 2018-09-09 by insomniacslk