u-root/u-root #933: acpidump command

rjoleary, 2018-09-14:


GanShun, 2018-09-14:
More information? I assume this is to dump the acpi tables from userspace?

rjoleary, 2018-09-14:
This is related to LinuxBoot:

We're having difficulty removing a few DXEs because they setup necessary ACPI tables. The high level solution is to follow these steps:

1. Run your original BIOS image and run `acpidump` to save to ACPI table to a file.
2. Perform dxectomy and include this ACPI file in the initramfs.
3. The boot kernel loads the acpitable from its initramfs.
4. Those troublesome DXEs can now be removed.

hugelgupf, 2019-02-08:
acpidump should use Linux kernel facilities to figure out where ACPI tables are, and implement a parser to read them. There's an equivalent non-Go acpidump out there.

The actual logic would likely be useful in a `pkg/acpi` to be reused by other things (like appending ACPI entries, etc).

See also https://01.org/linux-acpi/utilities