u-root/u-root #936: Vendoring is broken for system implementation

bluecmd, 2018-09-15:
Hi,

Trying to build u-bmc includes the u-root's vendor directory.
```
[bluecmd]$ fakeroot make                                                                                                  [u-bmc] [06:51:48]
go get github.com/u-root/u-root
go build -o u-root github.com/u-root/u-root
make -C cmd/uinit ssh_keys.go
make[1]: Entering directory '/home/bluecmd/go/src/github.com/u-root/u-bmc/cmd/uinit'
make[1]: 'ssh_keys.go' is up to date.
make[1]: Leaving directory '/home/bluecmd/go/src/github.com/u-root/u-bmc/cmd/uinit'
GOARM=5 GOARCH=arm ./u-root -build=bb -o initramfs.cpio \
	github.com/u-root/u-root/cmds/*/ \
	github.com/u-root/elvish \
	github.com/u-root/u-bmc/cmd/*/
2018/09/15 15:51:50 Build environment: GOARCH=arm GOOS=linux GOROOT=/usr/local/go GOPATH=/home/bluecmd/go CGO_ENABLED=0
2018/09/15 15:51:50 Filename is initramfs.cpio
2018/09/15 15:51:50 Skipping package "/home/bluecmd/go/src/github.com/u-root/u-root/cmds/.gitignore": cannot find package "." in:
	/home/bluecmd/go/src/github.com/u-root/u-root/cmds/.gitignore
2018/09/15 15:51:50 Skipping package "/home/bluecmd/go/src/github.com/u-root/u-bmc/cmd/.gitignore": cannot find package "." in:
	/home/bluecmd/go/src/github.com/u-root/u-bmc/cmd/.gitignore
2018/09/15 15:51:54 error building uroot.BuildOpts{Env:golang.Environ{Context:build.Context{GOARCH:"arm", GOOS:"linux", GOROOT:"/usr/local/go", GOPATH:"/home/bluecmd/go", CgoEnabled:false, UseAllFiles:false, Compiler:"gc", BuildTags:[]string(nil), ReleaseTags:[]string{"go1.1", "go1.2", "go1.3", "go1.4", "go1.5", "go1.6", "go1.7", "go1.8", "go1.9", "go1.10", "go1.11"}, InstallSuffix:"", JoinPath:(func(...string) string)(nil), SplitPathList:(func(string) []string)(nil), IsAbsPath:(func(string) bool)(nil), IsDir:(func(string) bool)(nil), HasSubdir:(func(string, string) (string, bool))(nil), ReadDir:(func(string) ([]os.FileInfo, error))(nil), OpenFile:(func(string) (io.ReadCloser, error))(nil)}}, Packages:[]string{"github.com/u-root/u-root/cmds/ansi", "github.com/u-root/u-root/cmds/basename", "github.com/u-root/u-root/cmds/boot", "github.com/u-root/u-root/cmds/boot2", "github.com/u-root/u-root/cmds/bzimage", "github.com/u-root/u-root/cmds/cat", "github.com/u-root/u-root/cmds/cbmem", "github.com/u-root/u-root/cmds/chmod", "github.com/u-root/u-root/cmds/chroot", "github.com/u-root/u-root/cmds/cmp", "github.com/u-root/u-root/cmds/comm", "github.com/u-root/u-root/cmds/console", "github.com/u-root/u-root/cmds/cp", "github.com/u-root/u-root/cmds/cpio", "github.com/u-root/u-root/cmds/date", "github.com/u-root/u-root/cmds/dd", "github.com/u-root/u-root/cmds/df", "github.com/u-root/u-root/cmds/dhclient", "github.com/u-root/u-root/cmds/dirname", "github.com/u-root/u-root/cmds/dmesg", "github.com/u-root/u-root/cmds/echo", "github.com/u-root/u-root/cmds/false", "github.com/u-root/u-root/cmds/field", "github.com/u-root/u-root/cmds/find", "github.com/u-root/u-root/cmds/free", "github.com/u-root/u-root/cmds/freq", "github.com/u-root/u-root/cmds/gpgv", "github.com/u-root/u-root/cmds/gpt", "github.com/u-root/u-root/cmds/grep", "github.com/u-root/u-root/cmds/gzip", "github.com/u-root/u-root/cmds/hexdump", "github.com/u-root/u-root/cmds/hostname", "github.com/u-root/u-root/cmds/id", "github.com/u-root/u-root/cmds/init", "github.com/u-root/u-root/cmds/insmod", "github.com/u-root/u-root/cmds/installcommand", "github.com/u-root/u-root/cmds/io", "github.com/u-root/u-root/cmds/ip", "github.com/u-root/u-root/cmds/kexec", "github.com/u-root/u-root/cmds/kill", "github.com/u-root/u-root/cmds/lddfiles", "github.com/u-root/u-root/cmds/ln", "github.com/u-root/u-root/cmds/losetup", "github.com/u-root/u-root/cmds/ls", "github.com/u-root/u-root/cmds/lsmod", "github.com/u-root/u-root/cmds/mkdir", "github.com/u-root/u-root/cmds/mkfifo", "github.com/u-root/u-root/cmds/mknod", "github.com/u-root/u-root/cmds/mktemp", "github.com/u-root/u-root/cmds/mm", "github.com/u-root/u-root/cmds/modprobe", "github.com/u-root/u-root/cmds/mount", "github.com/u-root/u-root/cmds/msr", "github.com/u-root/u-root/cmds/mv", "github.com/u-root/u-root/cmds/netcat", "github.com/u-root/u-root/cmds/ntpdate", "github.com/u-root/u-root/cmds/pci", "github.com/u-root/u-root/cmds/pe", "github.com/u-root/u-root/cmds/ping", "github.com/u-root/u-root/cmds/printenv", "github.com/u-root/u-root/cmds/ps", "github.com/u-root/u-root/cmds/pwd", "github.com/u-root/u-root/cmds/pxeboot", "github.com/u-root/u-root/cmds/readlink", "github.com/u-root/u-root/cmds/rm", "github.com/u-root/u-root/cmds/rmmod", "github.com/u-root/u-root/cmds/rsdp", "github.com/u-root/u-root/cmds/rush", "github.com/u-root/u-root/cmds/scp", "github.com/u-root/u-root/cmds/seq", "github.com/u-root/u-root/cmds/shutdown", "github.com/u-root/u-root/cmds/sleep", "github.com/u-root/u-root/cmds/sort", "github.com/u-root/u-root/cmds/sos", "github.com/u-root/u-root/cmds/srvfiles", "github.com/u-root/u-root/cmds/sshd", "github.com/u-root/u-root/cmds/strings", "github.com/u-root/u-root/cmds/stty", "github.com/u-root/u-root/cmds/switch_root", "github.com/u-root/u-root/cmds/sync", "github.com/u-root/u-root/cmds/tail", "github.com/u-root/u-root/cmds/tcz", "github.com/u-root/u-root/cmds/tee", "github.com/u-root/u-root/cmds/time_sos", "github.com/u-root/u-root/cmds/tr", "github.com/u-root/u-root/cmds/true", "github.com/u-root/u-root/cmds/truncate", "github.com/u-root/u-root/cmds/umount", "github.com/u-root/u-root/cmds/uname", "github.com/u-root/u-root/cmds/uniq", "github.com/u-root/u-root/cmds/unshare", "github.com/u-root/u-root/cmds/upspin_sos", "github.com/u-root/u-root/cmds/validate", "github.com/u-root/u-root/cmds/vboot", "github.com/u-root/u-root/cmds/wc", "github.com/u-root/u-root/cmds/wget", "github.com/u-root/u-root/cmds/which", "github.com/u-root/u-root/cmds/wifi", "github.com/u-root/u-root/cmds/xinit", "github.com/u-root/elvish", "github.com/u-root/u-bmc/cmd/assimilate", "github.com/u-root/u-bmc/cmd/fan", "github.com/u-root/u-bmc/cmd/freeze", "github.com/u-root/u-bmc/cmd/gpiowatcher", "github.com/u-root/u-bmc/cmd/socreset", "github.com/u-root/u-bmc/cmd/uinit"}, TempDir:"/tmp/u-root173712928/builder111116031", BinaryDir:"bbin"}: error building go package in "/home/bluecmd/go/src/github.com/u-root/u-root/bb": cmds/uinit/grpc.go:13:2: cannot find package "google.golang.org/grpc" in any of:
	/home/bluecmd/go/src/github.com/u-root/u-root/vendor/google.golang.org/grpc (vendor tree)
	/usr/local/go/src/google.golang.org/grpc (from $GOROOT)
	/home/bluecmd/go/src/google.golang.org/grpc (from $GOPATH)
cmds/uinit/grpc.go:14:2: cannot find package "google.golang.org/grpc/reflection" in any of:
	/home/bluecmd/go/src/github.com/u-root/u-root/vendor/google.golang.org/grpc/reflection (vendor tree)
	/usr/local/go/src/google.golang.org/grpc/reflection (from $GOROOT)
	/home/bluecmd/go/src/google.golang.org/grpc/reflection (from $GOPATH)
, exit status 1
Makefile:60: recipe for target 'initramfs.cpio' failed
make: *** [initramfs.cpio] Error 1
```

Note the printout:
```
	/home/bluecmd/go/src/github.com/u-root/u-root/vendor/google.golang.org/grpc (vendor tree)
```

bluecmd, 2018-09-15:
@hugelgupf

bluecmd, 2018-09-23:
Here is another one when I try to include `github.com/fullstorydev/grpcurl/cmd/grpcurl`:
```
GOARM=5 GOARCH=arm ./u-root \
	-build=bb \
	-o "initramfs.cpio.tmp" \
	core \
	github.com/u-root/u-root/cmds/scp/ \
	github.com/u-root/u-root/cmds/sshd/ \
	github.com/fullstorydev/grpcurl/cmd/grpcurl \
	github.com/u-root/u-bmc/cmd/*/ \
	github.com/u-root/u-bmc/platform/quanta-f06-leopard-ddr3/cmd/*/
2018/09/23 09:40:36 Build environment: GOARCH=arm GOOS=linux GOROOT=/usr/local/go GOPATH=/home/bluecmd/go CGO_ENABLED=0
2018/09/23 09:40:36 Filename is initramfs.cpio.tmp
2018/09/23 09:40:36 Skipping package "/home/bluecmd/go/src/github.com/u-root/u-bmc/cmd/.gitignore": cannot find package "." in:
	/home/bluecmd/go/src/github.com/u-root/u-bmc/cmd/.gitignore
2018/09/23 09:40:55 error building builder.Opts{Env:golang.Environ{Context:build.Context{GOARCH:"arm", GOOS:"linux", GOROOT:"/usr/local/go", GOPATH:"/home/bluecmd/go", CgoEnabled:false, UseAllFiles:false, Compiler:"gc", BuildTags:[]string(nil), ReleaseTags:[]string{"go1.1", "go1.2", "go1.3", "go1.4", "go1.5", "go1.6", "go1.7", "go1.8", "go1.9", "go1.10", "go1.11"}, InstallSuffix:"", JoinPath:(func(...string) string)(nil), SplitPathList:(func(string) []string)(nil), IsAbsPath:(func(string) bool)(nil), IsDir:(func(string) bool)(nil), HasSubdir:(func(string, string) (string, bool))(nil), ReadDir:(func(string) ([]os.FileInfo, error))(nil), OpenFile:(func(string) (io.ReadCloser, error))(nil)}}, Packages:[]string{"github.com/u-root/u-root/cmds/ansi", "github.com/u-root/u-root/cmds/boot", "github.com/u-root/u-root/cmds/cat", "github.com/u-root/u-root/cmds/cbmem", "github.com/u-root/u-root/cmds/chmod", "github.com/u-root/u-root/cmds/chroot", "github.com/u-root/u-root/cmds/cmp", "github.com/u-root/u-root/cmds/console", "github.com/u-root/u-root/cmds/cp", "github.com/u-root/u-root/cmds/cpio", "github.com/u-root/u-root/cmds/date", "github.com/u-root/u-root/cmds/dd", "github.com/u-root/u-root/cmds/df", "github.com/u-root/u-root/cmds/dhclient", "github.com/u-root/u-root/cmds/dirname", "github.com/u-root/u-root/cmds/dmesg", "github.com/u-root/u-root/cmds/echo", "github.com/u-root/u-root/cmds/elvish", "github.com/u-root/u-root/cmds/false", "github.com/u-root/u-root/cmds/field", "github.com/u-root/u-root/cmds/find", "github.com/u-root/u-root/cmds/free", "github.com/u-root/u-root/cmds/freq", "github.com/u-root/u-root/cmds/gpgv", "github.com/u-root/u-root/cmds/gpt", "github.com/u-root/u-root/cmds/grep", "github.com/u-root/u-root/cmds/gzip", "github.com/u-root/u-root/cmds/hexdump", "github.com/u-root/u-root/cmds/hostname", "github.com/u-root/u-root/cmds/id", "github.com/u-root/u-root/cmds/init", "github.com/u-root/u-root/cmds/insmod", "github.com/u-root/u-root/cmds/installcommand", "github.com/u-root/u-root/cmds/io", "github.com/u-root/u-root/cmds/ip", "github.com/u-root/u-root/cmds/kexec", "github.com/u-root/u-root/cmds/kill", "github.com/u-root/u-root/cmds/lddfiles", "github.com/u-root/u-root/cmds/ln", "github.com/u-root/u-root/cmds/losetup", "github.com/u-root/u-root/cmds/ls", "github.com/u-root/u-root/cmds/lsmod", "github.com/u-root/u-root/cmds/mkdir", "github.com/u-root/u-root/cmds/mkfifo", "github.com/u-root/u-root/cmds/mknod", "github.com/u-root/u-root/cmds/modprobe", "github.com/u-root/u-root/cmds/mount", "github.com/u-root/u-root/cmds/msr", "github.com/u-root/u-root/cmds/mv", "github.com/u-root/u-root/cmds/netcat", "github.com/u-root/u-root/cmds/ntpdate", "github.com/u-root/u-root/cmds/pci", "github.com/u-root/u-root/cmds/ping", "github.com/u-root/u-root/cmds/printenv", "github.com/u-root/u-root/cmds/ps", "github.com/u-root/u-root/cmds/pwd", "github.com/u-root/u-root/cmds/pxeboot", "github.com/u-root/u-root/cmds/readlink", "github.com/u-root/u-root/cmds/rm", "github.com/u-root/u-root/cmds/rmmod", "github.com/u-root/u-root/cmds/rsdp", "github.com/u-root/u-root/cmds/seq", "github.com/u-root/u-root/cmds/shutdown", "github.com/u-root/u-root/cmds/sleep", "github.com/u-root/u-root/cmds/sort", "github.com/u-root/u-root/cmds/stty", "github.com/u-root/u-root/cmds/switch_root", "github.com/u-root/u-root/cmds/sync", "github.com/u-root/u-root/cmds/tail", "github.com/u-root/u-root/cmds/tee", "github.com/u-root/u-root/cmds/true", "github.com/u-root/u-root/cmds/truncate", "github.com/u-root/u-root/cmds/umount", "github.com/u-root/u-root/cmds/uname", "github.com/u-root/u-root/cmds/uniq", "github.com/u-root/u-root/cmds/unshare", "github.com/u-root/u-root/cmds/validate", "github.com/u-root/u-root/cmds/vboot", "github.com/u-root/u-root/cmds/wc", "github.com/u-root/u-root/cmds/wget", "github.com/u-root/u-root/cmds/which", "github.com/u-root/u-root/cmds/scp", "github.com/u-root/u-root/cmds/sshd", "github.com/fullstorydev/grpcurl/cmd/grpcurl", "github.com/u-root/u-bmc/cmd/fan", "github.com/u-root/u-bmc/cmd/socreset", "github.com/u-root/u-bmc/platform/quanta-f06-leopard-ddr3/cmd/gpiowatcher", "github.com/u-root/u-bmc/platform/quanta-f06-leopard-ddr3/cmd/uinit"}, TempDir:"/tmp/u-root320674044/builder405414955", BinaryDir:"bbin"}: error building go package in "/home/bluecmd/go/src/github.com/u-root/u-root/bb": # github.com/u-root/u-root/bb/cmds/grpcurl
cmds/grpcurl/grpcurl.go:437:20: invalid operation: h.stat.Code() != "github.com/u-root/u-root/vendor/google.golang.org/grpc/codes".OK (mismatched types "google.golang.org/grpc/codes".Code and "github.com/u-root/u-root/vendor/google.golang.org/grpc/codes".Code)
, exit status 2
Makefile:83: recipe for target 'initramfs.cpio' failed
make: *** [initramfs.cpio] Error 1
```

hugelgupf, 2018-10-26:
So... this is still fixed by #936 for me, the u-bmc vendor tree just had to be fixed:

```
hugelgupf :: git fetch bluecmd
remote: Enumerating objects: 133, done.
remote: Counting objects: 100% (133/133), done.
remote: Compressing objects: 100% (82/82), done.
remote: Total 166 (delta 46), reused 133 (delta 46), pack-reused 33
Receiving objects: 100% (166/166), 374.07 KiB | 1.56 MiB/s, done.
Resolving deltas: 100% (48/48), completed with 19 local objects.
From https://github.com/bluecmd/u-bmc
 * [new branch]      correction    -> bluecmd/correction
 * [new branch]      dhcpv6        -> bluecmd/dhcpv6
 * [new branch]      flash-lock    -> bluecmd/flash-lock
 * [new branch]      master        -> bluecmd/master
 * [new branch]      re-vendor     -> bluecmd/re-vendor
 * [new branch]      verified-boot -> bluecmd/verified-boot
hugelgupf :: git checkout re-vendor
Branch 're-vendor' set up to track remote branch 're-vendor' from 'bluecmd'.
Switched to a new branch 're-vendor'
hugelgupf :: cd ../u-root
hugelgupf :: GOARM=5 GOARCH=arm ./u-root \                                                                                                                           u-root/u-root [master $%]
        -build=bb \
        -o "initramfs.cpio.tmp" \
        core \
        github.com/u-root/u-root/cmds/scp/ \
        github.com/u-root/u-root/cmds/sshd/ \
        github.com/u-root/u-bmc/cmd/\*/ \
        github.com/u-root/u-bmc/platform/quanta-f06-leopard-ddr3/cmd/\*/
2018/10/26 13:48:49 Disabling CGO for u-root...
2018/10/26 13:48:49 Build environment: GOARCH=arm GOOS=linux GOROOT=/usr/lib/google-golang GOPATH=/usr/local/google/home/chrisko/go CGO_ENABLED=0
2018/10/26 13:48:49 Filename is initramfs.cpio.tmp
2018/10/26 13:49:01 Successfully wrote initramfs.
```

bluecmd, 2018-10-26:
Confirmed, #992 should fix this.

bluecmd, 2018-11-05:
Reopening as integration tests seems to fail, and I don't *think* I'm doing something wrong on my side.

https://github.com/u-root/u-bmc/pull/96

hugelgupf, 2018-11-06:
Commented [here](https://github.com/u-root/u-bmc/pull/96#issuecomment-436100830) why it didn't work for the integration tests.

PR #992 (2018-10-25): bb: generate packages in place.
This means that when imported, each package will automatically use the
correct vendored dependencies in its subtree.

Previously:
Pkg github.com/u-bmc/u-bmc/cmds/foo would have files generated in
github.com/u-root/u-root/bb/cmds/foo. This meant that u-root's vendor
directory would be used, when the original foo was written with u-bmc's
vendor directory.

This way, generated package files are placed in
github.com/u-bmc/u-bmc/cmds/foo/.bb.

This hopefully fixes #936 for now.
closed 2018-11-06 by hugelgupf