u-root/u-root #944: Is UTF-8 the default? Implement case insensitive `uniq`

JohnAZoidberg, 2018-09-16:
The roadmap says `-i` is to be implemented for the `uniq` command.

The current implementation of `uniq` doesn't use the `string` type but `[]byte` to represent the lines.
Now the questions is: Do we expect the encoding to always be UTF-8? Then it's easy because `byte` has the `func ToLower(s []byte) []byte` function which interprets the bytes as UTF-8.

I'm no Go expert, having written only a couple hundred lines - I guess `[]bytes` is used because it's more perfomant because it doesn't need to worry about encoding and can reuse memory?

The coreutils implementation seems to treat the lines as ASCII.

DeDeRon, 2018-12-07:
there is bytes.Equalfold which makes the implementation trivial, but i wonder if this functionality is really needed. 

i also noticed that the u-root uniq implementation takes 1..n files as argument. gnu uniq and plan 9 uniq take only one file, which makes more sense because it's is less confusing. if someone want's to uniq more than one file using "cat file1, file2.. | uniq" is a viable option.

hugelgupf, 2019-02-08:
We've actually also been talking about Unicode through a different issue: the unicode package adds a whole bunch of tables of characters to the binary size of a u-root program. It might bring us ~500K of savings to implement a few ASCII-only equivalents of string functions -- strings.ToLower, strings.ToUpper, etc, but ASCII-only.