u-root/u-root #954: bb builder fails to create lock files if the hard links crosses file systems

insomniacslk, 2018-09-21:
The busybox mode uses a library (github.com/nightlyone/lockfile) that creates hardlinks in order to do locking. If the source and destination of the link are on different file systems, this of course fails (hard links cannot span across filesystems), and it's breaking the build for us under those conditions.

It's a strong assumption that a hard link can always be done - what about switching to a locking library that uses a different approach?

insomniacslk, 2018-09-21:
cc @tfg13 @Natolumin

hugelgupf, 2018-09-21:
Lolwat. Sorry about that, I thought that locking library did something sane.

insomniacslk, 2018-09-21:
no worries! As discussed on Slack let's switch to something saner :) or let's ask them to use symlinks on non-windows platforms, either works for us

insomniacslk, 2018-09-21:
I've opened an issue (and will likely send a PR) to the `lockfile` library. Otherwise I guess we should switch away to another library or write it ourselves

closed 2018-10-25 by hugelgupf