u-root/u-root #985: binary build mode fails when including commands with vendor dependencies

ddiss, 2018-10-18:
Generating an image with "binary" build mode fails:
```
git log  --pretty=oneline --abbrev-commit   -1
8ebf81a3 (HEAD -> master, origin/master, origin/HEAD) initramfs: now that we follow symlinks, don't lstat.

./u-root -defaultsh rush -build binary -o ./myinitrd github.com/u-root/u-root/cmds/{init,ls} github.com/u-root/u-root/xcmds/rush
2018/10/18 15:44:58 Disabling CGO for u-root...
2018/10/18 15:44:58 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib64/go/1.11 GOPATH=/home/ddiss/go:/usr/share/go/1.11/contrib CGO_ENABLED=0
2018/10/18 15:44:58 Filename is ./myinitrd
2018/10/18 15:45:01 error building builder.Opts{Env:golang.Environ{Context:build.Context{GOARCH:"amd64", GOOS:"linux", GOROOT:"/usr/lib64/go/1.11", GOPATH:"/home/ddiss/go:/usr/share/go/1.11/contrib", CgoEnabled:false, UseAllFiles:false, Compiler:"gc", BuildTags:[]string(nil), ReleaseTags:[]string{"go1.1", "go1.2", "go1.3", "go1.4", "go1.5", "go1.6", "go1.7", "go1.8", "go1.9", "go1.10", "go1.11"}, InstallSuffix:"", JoinPath:(func(...string) string)(nil), SplitPathList:(func(string) []string)(nil), IsAbsPath:(func(string) bool)(nil), IsDir:(func(string) bool)(nil), HasSubdir:(func(string, string) (string, bool))(nil), ReadDir:(func(string) ([]os.FileInfo, error))(nil), OpenFile:(func(string) (io.ReadCloser, error))(nil)}}, Packages:[]string{"github.com/u-root/u-root/cmds/init", "github.com/u-root/u-root/cmds/ls", "github.com/u-root/u-root/xcmds/rush"}, TempDir:"/tmp/u-root542580345/builder429597060", BinaryDir:"bin"}: error building go package in "/home/ddiss/go/src/github.com/u-root/u-root/cmds/ls": ls.go:27:2: cannot find package "github.com/spf13/pflag" in any of:
        /usr/lib64/go/1.11/src/github.com/spf13/pflag (from $GOROOT)
        /home/ddiss/go/src/github.com/spf13/pflag (from $GOPATH)
        /usr/share/go/1.11/contrib/src/github.com/spf13/pflag
, exit status 1
```
The missing _github.com/spf13/pflag_ dependency is present in the top level vendor dir:
```
ls -d vendor/github.com/spf13/pflag
vendor/github.com/spf13/pflag
```

Using "bb" build mode works fine:
```
./u-root -defaultsh rush -build bb -o ./myinitrd github.com/u-root/u-root/cmds/{init,ls} github.com/u-root/u-root/xcmds/rush2018/10/18 15:41:47 Disabling CGO for u-root...
2018/10/18 15:41:47 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib64/go/1.11 GOPATH=/home/ddiss/go:/usr/share/go/1.11/contrib CGO_ENABLED=0
2018/10/18 15:41:47 Filename is ./myinitrd
2018/10/18 15:41:52 Successfully wrote initramfs.
```

rjoleary, 2018-10-22:
@hugelgupf is working on a fix.

hugelgupf, 2018-10-25:
Actually looks like an error independent from myinitrd:

```go
error building go package in "/home/ddiss/go/src/github.com/u-root/u-root/cmds/ls": ls.go:27:2: cannot find package "github.com/spf13/pflag" in any of:
        /usr/lib64/go/1.11/src/github.com/spf13/pflag (from $GOROOT)
        /home/ddiss/go/src/github.com/spf13/pflag (from $GOPATH)
        /usr/share/go/1.11/contrib/src/github.com/spf13/pflag
, exit status 1
```

But I haven't been able to reproduce locally. Did you perhaps turn on the go mod experiment with Go 1.11?
This is what I have:

```bash
$ go version
go version go1.11rc2 linux/amd64

$ ./u-root -defaultsh=rush -build=binary ./cmds/{ls,init} ./xcmds/rush
2018/10/24 20:02:49 Disabling CGO for u-root...
2018/10/24 20:02:49 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib/google-golang GOPATH=/usr/local/google/home/chrisko/go CGO_ENABLED=0
2018/10/24 20:02:49 Filename is /tmp/initramfs.linux_amd64.cpio
2018/10/24 20:02:52 Successfully wrote initramfs.
```

ddiss, 2018-10-25:
The go mod experiment isn't enabled, AFAICT.
I'm using a downstream go1.11.1 build (https://build.opensuse.org/package/show/devel:languages:go/go1.11), which has a few minor patches. I'll retry with an upstream build when I get a spare moment. Feel free to close as WORKSFORME.
```
ddiss@echidna:~/go/src/github.com/u-root/u-root (master)> env|grep GO
GOARCH=amd64
GOROOT=/usr/lib64/go/1.11
GOOS=linux
GOPATH=/home/ddiss/go:/usr/share/go/1.11/contrib
ddiss@echidna:~/go/src/github.com/u-root/u-root (master)> go build u-root.go 
ddiss@echidna:~/go/src/github.com/u-root/u-root (master)> git log --pretty=oneline -1 |cat
c3995797780d4d223e496b85e3ae0a27d43da4df add prettyprinting
ddiss@echidna:~/go/src/github.com/u-root/u-root (master)> go build u-root.go 
ddiss@echidna:~/go/src/github.com/u-root/u-root (master)> go version
go version go1.11.1 linux/amd64
ddiss@echidna:~/go/src/github.com/u-root/u-root (master)> ./u-root -defaultsh=rush -build=binary ./cmds/{ls,init} ./xcmds/rush
2018/10/25 11:58:01 Disabling CGO for u-root...
2018/10/25 11:58:01 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib64/go/1.11 GOPATH=/home/ddiss/go:/usr/share/go/1.11/contrib CGO_ENABLED=0
2018/10/25 11:58:01 Filename is /tmp/initramfs.linux_amd64.cpio
2018/10/25 11:58:04 error building builder.Opts{Env:golang.Environ{Context:build.Context{GOARCH:"amd64", GOOS:"linux", GOROOT:"/usr/lib64/go/1.11", GOPATH:"/home/ddiss/go:/usr/share/go/1.11/contrib", CgoEnabled:false, UseAllFiles:false, Compiler:"gc", BuildTags:[]string(nil), ReleaseTags:[]string{"go1.1", "go1.2", "go1.3", "go1.4", "go1.5", "go1.6", "go1.7", "go1.8", "go1.9", "go1.10", "go1.11"}, InstallSuffix:"", JoinPath:(func(...string) string)(nil), SplitPathList:(func(string) []string)(nil), IsAbsPath:(func(string) bool)(nil), IsDir:(func(string) bool)(nil), HasSubdir:(func(string, string) (string, bool))(nil), ReadDir:(func(string) ([]os.FileInfo, error))(nil), OpenFile:(func(string) (io.ReadCloser, error))(nil)}}, Packages:[]string{"github.com/u-root/u-root/cmds/ls", "github.com/u-root/u-root/cmds/init", "github.com/u-root/u-root/xcmds/rush"}, TempDir:"/tmp/u-root333496313/builder534728964", BinaryDir:"bin"}: error building go package in "/home/ddiss/go/src/github.com/u-root/u-root/xcmds/rush": tty_linux.go:14:2: cannot find package "golang.org/x/sys/unix" in any of:
        /usr/lib64/go/1.11/src/golang.org/x/sys/unix (from $GOROOT)
        /home/ddiss/go/src/golang.org/x/sys/unix (from $GOPATH)
        /usr/share/go/1.11/contrib/src/golang.org/x/sys/unix
, exit status 1
```

hugelgupf, 2019-02-08:
Hey @ddiss, does this still happen for you?

ddiss, 2019-02-09:
hi @hugelgupf , yes I still hit this:

```
~/go/src/github.com/u-root/u-root (master)> go version
go version go1.11.5 linux/amd64

~/go/src/github.com/u-root/u-root (master)> git log --pretty=oneline -1|cat
28fed39e348d1e3b385289f4a9812ecd47f6fe78 Changes to make required DCO clearer

~/go/src/github.com/u-root/u-root (master)> ./u-root -defaultsh rush -build binary -o ./myinitrd github.com/u-root/u-root/cmds/{init,ls} github.com/u-root/u-root/xcmds/rush
2019/02/09 17:54:41 Disabling CGO for u-root...
2019/02/09 17:54:41 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib64/go/1.11 GOPATH=/home/ddiss/go CGO_ENABLED=0
2019/02/09 17:54:41 Filename is ./myinitrd
2019/02/09 17:54:44 error building: error building go package in "/home/ddiss/go/src/github.com/u-root/u-root/xcmds/rush": tty_linux.go:14:2: cannot find package "golang.org/x/sys/unix" in any of:
        /usr/lib64/go/1.11/src/golang.org/x/sys/unix (from $GOROOT)
        /home/ddiss/go/src/golang.org/x/sys/unix (from $GOPATH)
, exit status 1

~/go/src/github.com/u-root/u-root (master)> l -d vendor/golang.org/x/sys/unix
drwxr-xr-x 1 ddiss users 10752 Feb  8 00:44 vendor/golang.org/x/sys/unix/

~/go/src/github.com/u-root/u-root (master)> ./u-root -defaultsh rush -build bb -o ./myinitrd github.com/u-root/u-root/cmds/{init,ls} github.com/u-root/u-root/xcmds/rush
2019/02/09 17:55:20 Disabling CGO for u-root...
2019/02/09 17:55:20 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib64/go/1.11 GOPATH=/home/ddiss/go CGO_ENABLED=0
2019/02/09 17:55:20 Filename is ./myinitrd
2019/02/09 17:55:24 Successfully wrote initramfs.
```

closed 2024-02-27 by hugelgupf