u-root/u-root #995: integrate alpine apk as replacement / alternative to tiny core tcz?

pwFoo, 2018-10-27:
I like alpine linux base / package manager and would prefer apk instead of tcz.
Why not move to alpine base or alpine packages?

bluecmd, 2018-11-03:
I don't think u-root wants to have an opinion here - if somebody wants it and wants to maintain it, adding an `apk` cmd should be fine. That way it's up to the folks that want it to link it in.

However, just like `tcz` I don't think `apk` would end up in the list of default packages as u-root by default is a self contained system.

pwFoo, 2018-11-04:
There is a apk-static package. So maybe instead of convert it to go it could be included / downloaded as binary if needed. I'll test it that way.

hugelgupf, 2018-11-09:
I've been thinking about pulling in Arch Linux packages (just tarballs, no squashfs necessary). If I can download & unpack apk packages using existing tools (without having to write a squashfs parser), I'd consider that, too :)

pwFoo, 2018-11-09:
I'll build a test with apk.static to see if it would run without additional dependencies.
https://pkgs.alpinelinux.org/contents?branch=edge&name=apk-tools-static&arch=x86&repo=main

I think apk packages could be just tar archives

hugelgupf, 2019-02-08:
I believe @zaolin has been working on some alpine stuff?