u-root/u-root #998: Searching for xinit /wifi documentation examples?

pwFoo, 2018-10-29:
Haven't found how to use u-root wifi or xinit. Is there documentation or an example available?

Found that PR for xinit #456. So it's possible to run chrome with u-root and xinit?

bluecmd, 2018-11-03:
Maybe you've already seen it, but if not have a look at https://github.com/u-root/NiChrome - it's probably close to what you're after. u-root itself is more like a framework, it doesn't have all the pieces for a full system, that's why we have NiChrome, SystemBoot, u-bmc etc that uses u-root but adds the system "around" it.

bluecmd, 2018-11-04:
Yes, you're 100% right in that documentation isn't in an excellent state. The upside is that the cmds are not huge, so reading the code is currently the best way to figure out what is actually happening.

rminnich, 2018-12-06:
Go comes from the Bell Labs tradition. The goal is clear, readable code that you don't have to document up, down, and sideways. We want to strive for that as well. At the very least you need to learn the language, you won't get far otherwise. 

The problem (seen in practice) with too many comments is that they tend to get stale. It's best if the code is readable enough not to need them. 

xinit is 63 commands. It's quite simple. 
wifi is 146 lines and there's not much to it. 
First step is do the go tour, then please take another look.
See Rob Pike's 2010 talk on the rationale for Go and the importance of readable code, then please fix the code if it's not readable enough.
Thanks.

closed 2018-12-06 by rminnich