u-root/u-root #1004: set cgroup use_hierarchy

pwFoo, 2018-10-31:
Docker(-runc) wants it:
```
[ 1160.858612] cgroup: docker-runc (359) created nested cgroup for controller "memory" which has incomplete hierarchy support. Nested cgroups may change behavior in the future.
[ 1160.865449] cgroup: "memory" requires setting use_hierarchy to 1 on the root
```

Fix
```
echo 1 > /sys/fs/cgroup/memory.use_hierarchy
```

Should be added to u-root init?

rminnich, 2018-12-05:
can you tell me more about what this means? We can easily put this in init.

pwFoo, 2018-12-05:
I just tested to run docker containers with u-root and the warning above was shown.  I'm not an cgroup expert, tested it by set it runtime and it looks fine. 

Found an linuxkit issue about that:
https://github.com/linuxkit/linuxkit/issues/373

Activate it should improve container / cgroup support?

rminnich, 2018-12-06:
One of the goals of this project is that we understand why we do something, rather than just doing it. It would be very helpful if you could find out more about what this means before we go blindly setting it. It did not used to be needed, now it does. Why? What does it do? Can you find some docs? That would be very helpful. Thanks for diagnosing this problem.

rminnich, 2019-06-07:
can you tell us which version of the kernel this happened on for you.
Here is some history:
ed3d261b53f51 kernel/cgroup.c        (Joe Perches                2014-04-25 18:28:03 -0400 4878)                pr_warn("%s (%d) created nested cgroup for controller \"%s\" which has incomplete hierarchy support. Nested cgroups may change behavior in the future.\n",
ed3d261b53f51 kernel/cgroup.c        (Joe Perches                2014-04-25 18:28:03 -0400 4881)                        pr_warn("\"memory\" requires setting use_hierarchy to 1 on the root\n");

So this stuff is from 5 and 6 years ago, and it's still a problem.

but what happens "in the future"? We end up with this bit of code forever. I'd like to be a bit more sure that we should drop this into u-root. In 3-4 years, will it be there and we won't know why?

pwFoo, 2019-06-07:
Hi @rminnich, I don't have the code from and haven't tested it in the last time. So if nobody need it, we can close that issue for now?

rjoleary, 2019-06-26:
I'll close it for now. Feel free to re-open when you come across the issue again!

closed 2019-06-26 by rjoleary