u-root/u-root #1054: Refactor boot/boot2 commands

hugelgupf, 2018-12-20:
These should really be one and the same command.

Their common code should go into pkg/diskboot, a library of GRUB / syslinux / LILO config parsers.

We should only support the simplest configs of those three systems. Complicated stuff? No go.

See also #751.

hugelgupf, 2019-05-13:
Existing commands:

https://github.com/u-root/u-root/tree/master/cmds/boot (has a syslinux config parser, and a GRUB config parser?)
https://github.com/u-root/u-root/tree/master/cmds/boot2 (has a GRUB config parser?)

The real logic for these should be reusable in a package.

Relevant packages:

https://github.com/u-root/u-root/tree/master/pkg/boot (generic tools)
https://github.com/u-root/u-root/tree/master/pkg/diskboot (GRUB parser??)
https://github.com/u-root/u-root/tree/master/pkg/pxe (has a syslinux config parser that should probably be broken out into its own package and made usable by pkg/pxe and pkg/diskboot)

Relevant existing PR:

https://github.com/u-root/u-root/pull/1155 (deleting code I wrote that never ended up being used)

hugelgupf, 2019-05-13:
Additionally, the GRUB and syslinux parsers should get support for multiboot kernels using https://github.com/u-root/u-root/tree/master/pkg/multiboot (and it's stubbed out in https://github.com/u-root/u-root/blob/master/pkg/boot/multiboot.go right now...)

rjoleary, 2021-06-07:
It looks like this is done since there's no more boot2.

closed 2021-06-07 by rjoleary