u-root/u-root #1063: kexec multiboot: unable to load modules

jensdrenhaus, 2019-01-07:
trying to boot *fedora* via *tboot* ([GitHub](http://github.com)) using *kexec* fails: 
[image]
Afterwards the machine reboots.

When booting the same *tboot* binary with the same kernel and initramfs as modules like this:
/boot/grub2/grub.cfg
``` 
menuentry ' Fodora with tboot'
      set root='hd0,msdos1'
      multiboot      /tboot.gz logging=serial,memory,vga
      module         /vmlinuz-4.18.16-300.fc29.x86_64 root=/dev/mapper/fedora-root ro # ... rest of cmdline
      module         /nitramfs-4.18.16-300.fc29.x86_64.img
```
then *tboot* output during startup looks like that and *fedora* starts successfully:
[image]

It looks to me that the modules are not loaded properly so that *tboot* connot access them.

In *uroot* I use kexec as follows:
[image]
with the following output (don't be confused, I've added some extra output in the code, starteing with DEBUG: ...)
[image]

n-canter, 2019-01-08:
I see one suspicious thing in your output: the modules' starts are not page aligned.
Multiboot spec doesn't require it but looks like many implementations do. I'll write a fix.

n-canter, 2019-01-09:
Fixed by #1064

closed 2019-01-09 by n-canter