u-root/u-root #1095: GPT block size

n-canter, 2019-01-24:
Determine the block size of disk.
Now gpt package assumes BlockSize is always 512B.

hugelgupf, 2019-02-08:
gpt here is `pkg/gpt`.

Block sizes can be something other than 512B. I believe there should be some metadata somewhere to tell us?

n-canter, 2019-02-19:
We can use ioctl(2) to get BlockSize. The only thing I don't understand which one do we need: physical block size or a logical one?

rminnich, 2019-03-19:
gpt notionally has all kinds of block sizes but in the many years of the standard's existence, it seems everyone assumes 512B, which is why I did not bother with figuring it out. Reading the standard, it seems that as usual with most of these standards, things were not well thought out.