u-root/u-root #1096: Cannot run u-root in QEMU if compiled in source mode

n-canter, 2019-01-24:
In bb mode everything works fine.
go was downloaded from official site as a binary release.
```
$ go version
go version go1.11.1 linux/amd64

$ ldd `which go`
	linux-vdso.so.1 (0x00007ffd8199a000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ff7a7e4b000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff7a7aac000)
	/lib64/ld-linux-x86-64.so.2 (0x00007ff7a8068000)

$ qemu-system-x86_64  --version
QEMU emulator version 3.1.0 (Debian 1:3.1+dfsg-2)
Copyright (c) 2003-2018 Fabrice Bellard and the QEMU Project developers
```
Kernel was built with the latest [config](https://github.com/u-root/u-root/blob/bdeb3a88958fc1020752dd347d2943ef1a7eae84/.circleci/images/test-image-amd64/config_linux4.17_x86_64.txt) from CircleCI.

```
$ u-root -o ./initramfs.linux_amd64.cpio all 
$ qemu-system-x86_64 -m 8192 -kernel ./bzImage -initrd ./initramfs.linux_amd64.cpio -nographic --append "console=ttyS0"
```
QEMU output:
```
...
2019/01/24 22:00:17 Couldn't compile "init": error building go package in "/src/github.com/u-root/u-root/cmds/init": # net
/go/src/net/lookup_unix.go:80:24: undefined: cgoLookupHost
/go/src/net/lookup_unix.go:95:24: undefined: cgoLookupIP
/go/src/net/lookup_unix.go:107:23: undefined: cgoLookupPort
/go/src/net/lookup_unix.go:123:24: undefined: cgoLookupCNAME
/go/src/net/lookup_unix.go:323:23: undefined: cgoLookupPTR
, exit status 2
Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100

Kernel Offset: 0x38e00000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100
 ]---

```

rjoleary, 2019-01-25:
The problem might be that Go is downloaded as a binary release. Anecdotally, I've built Go1.11 from source and u-root source mode runs fine in qemu.

One possibility that Ron suggested is the Go compiler might behave differently if it was originally built with CGO_ENABLED=0. I built my compiler with CGO_ENABLED=0, so that's one data point.

Unfortunately, we do not have automated tests for source mode. I tried to set them up in #848, but running them under qemu in circleci (which does not support kvm) was far too slow and times out.

hugelgupf, 2019-02-08:
This was fixed by #1128, I believe.

closed 2019-02-08 by hugelgupf