u-root/u-root #1179: busybox should not panic on trivial errors like identical names

rminnich, 2019-04-05:
If there are two packages that try to register the same command name, busybox panics. This is bad behavior for firmware. It should print a diagnostic and add the package with a mangled or unique name.

hugelgupf, 2019-04-05:
I think it should panic, cause it's a build system programming error. Or how would you choose which of the two to run?

rminnich, 2019-04-05:
probably uniquify them. The thing is, with firmware, it's best to not leave it in a dead state.

jonzhang-fb, 2019-04-05:
Some questions:

  1.  Why not let the first package who registered the command name to win? And the subsequent packages fail to register?
  2.  Can this (command name conflict) be caught at build time, when building “busybox”?

jonzhang-fb, 2019-04-05:
I take my questions back. I realized that Ron had opened issue 1178 and 1179 to address both of them.

closed 2020-09-24 by hugelgupf