u-root/u-root #1200: dhclient: ability to unicast to a server

hugelgupf, 2019-04-23:
Currently, DHCPv4 will use a connection that by default broadcasts packets on the network when it sends DHCPDISCOVER and DHCPREQUEST messages.

We have some use cases where we'd rather establish a direct connection to a DHCP server.

This *should* be as simple as swapping out the `net.PacketConn` used by `nclient4` with one derived from a `net.ListenUDP`

the user for this is @sdab, please work with him

hugelgupf, 2019-04-24:
I was hoping `dhclient` would gain a command-line flag like `dhclient -server 192.168.0.10` which if used creates the different connection.

pallaud, 2019-05-22:
I had a couple questions to make sure I'm understanding this right - do we want to unicast because clients that already have established IP addresses are requesting renewal of the lease? If so then rather than building the DHCPv4 request from the given offer, would we just use the server address given by the dhclient flag and change the connection type to `net.UDPConn`?

closed 2024-02-20 by hugelgupf