u-root/u-root #1209: Integration test TestMultiboot is flaky

rjoleary, 2019-05-09:
The error is:

```
    --- FAIL: TestMultiboot//kernel.gz (7.79s)
        integration.go:113: Filename is /tmp/uroot-integration856086765/initramfs.cpio
        integration.go:113: No default shell: command or path "elvish" not included in u-root build
        integration.go:170: QEMU command line for testMultiboot:
            /home/circleci/qemu-system-x86_64 -L /home/circleci/pc-bios -m 1G -nographic -kernel /tmp/uroot-integration856086765/bzImage -append 'console=ttyS0 earlyprintk=ttyS0' -initrd /tmp/uroot-integration856086765/initramfs.cpio -drive file=fat:rw:/tmp/uroot-integration856086765,if=none,id=tmpdir -device ich9-ahci,id=ahci -device ide-drive,drive=tmpdir,bus=ahci.0 -device virtio-rng-pci
        multiboot_test.go:61: Cannot unmarshal multiboot information from executed kernel: unexpected end of JSON input
        integration.go:176: keeping temp dir:  /tmp/uroot-integration856086765
```

mpictor, 2019-05-09:
Looks like you read from qemu's stdout?

Several years ago I wrote some kernel module unit tests and ran them in a vm. I used a virtual fs, with tests writing files out. Once all tests ran I exec'd `shutdown -h now`, and once qemu shut down I read the files written by the tests. It was a bit of a pain to set up, but I don't recall it being flaky. Might be worth investigating...

qemu args:
```
-fsdev local,id=sharedrv,path=$testdir,security_model=mapped-file
-device virtio-9p-pci,fsdev=sharedrv,mount_tag=/dev/root
```

kernel args:
```
devtmpfs.mount=1 rw root=/dev/root rootfstype=9p rootflags=trans=virtio,version=9p2000.L
```

I don't recall exactly why I used `security_model=mapped-file`, but it did require creating attr files so that init (etc) were executable. Annoyingly, I don't see any examples of what the attr file should contain, just my notes that it wasn't obvious. Poking at qemu's source might be necessary, and/or creating a file in the vm, setting attributes, then looking at the file on the host.

rjoleary, 2019-05-14:
There's not much to go on to debug this one. You're right that it seems the serial was cut-off early and the JSON was unparsable. Perhaps the timeout can be increase. At least we should figure out why the serial logs are missing.

As for using 9p, that's better than what we're doing right now. What we have right now is:

* The input to QEMU is a bunch of files passed in via read-only FAT over ahci emulation. I couldn't for the life of me figure out how to get this to work with qemu+arm, so a bunch of integration tests are amd64-only. :disappointed: 
* The output from QEMU is serial. Depending on the test, it is encoded in different ways.

rjoleary, 2019-05-14:
I see why this is flaky. A `bytes.Buffer` is passed as `qemu.Options.SerialOutput` here: https://github.com/u-root/u-root/blob/master/integration/multiboot_test.go#L18 , which in turn is passed to `goexpect.Tee`.

A `bytes.Buffer` is not thread-safe while the `goexpect.Tee` function expects a thread-safe writer as stated in the documentation here: https://github.com/u-root/u-root/blob/master/vendor/github.com/google/goexpect/expect.go#L104 . This should probably be documented in the `qemu` package as well.

Possibly, one solution is to use `io.PipeWriter`.

mpictor, 2019-05-20:
> The input to QEMU is a bunch of files passed in via read-only FAT over ahci emulation. I couldn't for the life of me figure out how to get this to work with qemu+arm, so a bunch of integration tests are amd64-only.

Assuming I found the right location ([devices.go:79](https://github.com/u-root/u-root/blob/master/pkg/qemu/devices.go#L79)), have you tried without `if=...` and `id=...`, and the corresponding device entries?

I'm pretty sure you can do `-drive file=fat:/path/to/dir` and qemu will chose the hardware to simulate to make it happen.

However, this assumes that qemu+arm fails because ich8-achi isn't supported on arm, either by qemu or by the kernel - and that qemu will choose something better.

mpictor, 2019-06-07:
If you do decide to use `9p` with `security-model=mapped-file`, your `/init` _must_ be owned by root - it's not enough for it to be executable by root.

Assuming `work` is the dir you share with the vm, this file is enough:
$ `cat work/.virtfs_metadata/init`
```
virtfs.uid=0
virtfs.gid=0
virtfs.mode=33261
```
33261 decimal -> 100755 octal

rjoleary, 2019-07-13:
Thanks for looking into using 9p! I think replacing vfat with 9p would be a great idea. Feel free to work on it and send a PR of you're eager!

I'm not so sure about replacing the initramfs with 9p because the initramfs is the format a lot of people use for LinuxBoot. It might be worth having tests using the 9p root in addition to the existing tests using the initramfs root.

I'm working on a small change to fix a race condition in how some of the tests do their logging, but it shouldn't interfere with adding 9p.

rminnich, 2019-07-13:
I'm not sold on 9p for a few reasons. At some point the Linux folks broke the kernel 9p client in a subtle way (again) such that it no longer correctly works with all 9p servers (I wrote one of the ones it does not work with, which has worked fine for four years for plan 9 but fails on linux because ... linux people still don't understand 9p ... they keep screwing it up). Further, I expect you'll see a big performance hit if you are using 9p for root file system as compared to a simple initramfs (I've tried this before and would need to see numbers showing me I'm wrong but I'm willing to be convinced). Finally, it's just a more complex setup and I'd still like to be convinced it's worth it. Sorry to be a wet blanket :-(

mpictor, 2019-07-15:
The use case I was intending to address is testing (not production), though in hindsight my comments on Slack were about _replacing_ cpio/initramfs rather than supplementing. Somehow I managed to forget that the cpio/initramfs code I was looking at is needed for more than just testing :confounded: 

@rminnich are your performance concerns only with production, or also for testing? I'm not sure it's even possible to boot off of _networked_ 9p in place of an initramfs. If it was, yes, I can see there being performance and/or reliability issues.

For testing, if we confine ourselves to using qemu's 9p-virtio as the server, I don't believe we'll have compatibility concerns; the only troubles I recall encountering were of the PEBKAC form. I _suspect_ that 9p will be faster than qemu's vvfat, and it certainly lacks many of the limitations vvfat has.

I've not tried 9p-virtio with anything but x86, but perhaps it works on arches that vvfat doesn't.

GanShun, 2019-07-15:
I think in production, since we have to use an initramfs, we would like the test to closely mirror that. I do like replacing the vfat with 9p though, I think that would be really great!

mpictor, 2019-07-16:
created PR for 9p: https://github.com/u-root/u-root/pull/1291

rminnich, 2019-07-16:
ok if you all are wanting to do it I remove my objections.

JulienVdG, 2019-11-05:
It seams we have new occurence of this here today: https://github.com/u-root/u-root/pull/1423 it looks to me that we have a race with goexpect... which is expected:
We wait for the string `"status": "ok"` (in integration/multiboot_test.go) and the log stops exactly at `"status": "ok"` !!
We could make it less racy by using the `expect.PartialMatch(true)` option, that would make the buffer to always stop just after the last match (allowing next match to start looking from here).
But in that case this would make it always fail.

In my opinion there is two missing parts in the test:
1. Match something including the closing JSON bracket (or something after it if any)
2. Ensure the buffer is filled by calling the `cleanup()` function before using the `serial` buffer.

JulienVdG, 2019-11-05:
It seams I'm partially wrong about PartialMatch, this only affect the internal matching buffer, not the Tee, we only need it when matching several successive regex.

Also `cleanup()` is probably not needed as if the match got the regex the strings were already send to Tee (= serial)

trying a fix in: https://github.com/u-root/u-root/pull/1427

closed 2019-11-05 by hugelgupf