u-root/u-root #1212: diskboot: BLS support

hugelgupf, 2019-05-13:
from @flammit

> noticed your updates to #1054. just wanted to note that it’s probably worth adding support / tests for parsing BLS configs now that Fedora 30 ships with it by default: https://fedoraproject.org/wiki/Changes/BootLoaderSpecByDefault

siro20, 2019-06-28:
localboot has the same problem, starting with fedora 30 it isn't able to find the boot entries in grub.cfg

tfg13, 2019-06-28:
There was also https://github.com/systemboot/systemboot/issues/95

I have a half-baked BLS implementation, I can try to get it somewhere in the next couple of days.

I just checked, it looks like the BLS impl in Fedora has the same issue as the RHEL8 one: One needs to parse the BLS files first, but also still parse some grub stuff, because BLS references variables defined there (mostly `$kernelopts`)

tfg13, 2019-07-09:
Stuff is happening in #1283

closed 2024-02-20 by hugelgupf