u-root/u-root #1214: modprobe fails inside u-root shell

srcUp, 2019-05-14:
I have built an initramfs using "files" option to include /lib/modules.
```
./u-root -format=cpio -build=source -files /lib/modules/`uname -r` -o initramfs-sluinit-files.cpio ./cmds/* examples/sluinit
```
After booting into u-root, I do see "/lib/modules/4.14.35-1838.el7uek.x86_64" folder populated,
but modprobe fails for most of the drivers.
```
~/> modprobe ata_generic
2019/05/14 23:28:14 modprobe: Could not load module "ata_generic": 
could not find module path "ata_generic": Could not find path for module "ata_generic"
Exception: modprobe exited with 1
[tty], line 1: modprobe ata_generic
```
lsmod doesn't print anything.
```
~/> lsmod
Module                  Size  Used by
```
Note that modules under /lib/modules/4.14.35-1838.el7uek.x86_64 are all compressed,
both on host kernel and the u-root shell.
```
ls /lib/modules/4.14.35-1838.el7uek.x86_64/kernel/drivers/bluetooth
ath3k.ko.xz    bfusb.ko.xz        bpa10x.ko.xz   btbcm.ko.xz    btmrvl.ko.xz       btrtl.ko.xz   btuart_cs.ko.xz  dtl1_cs.ko.xz   hci_vhci.ko.xz
bcm203x.ko.xz  bluecard_cs.ko.xz  bt3c_cs.ko.xz  btintel.ko.xz  btmrvl_sdio.ko.xz  btsdio.ko.xz  btusb.ko.xz      hci_uart.ko.xz
```

Configs:
```
Kernel: 4.14.35-1838.el7uek.x86_64
qemu-system-x86_64: qemu-3.1.0-2.el7
u-root: https://github.com/TrenchBoot/u-root

rjoleary, 2019-05-15:
It seems like the responsibility for decompressing modules falls on user-mode which our insmod does not yet support.

Would it work if you build the initramfs like this instead?

```
cp -RLp /lib/modules/`uname -r` /tmp/modules
unxz `find -name '*.ko.xz' /tmp/modules`
./u-root -format=cpio -build=source -files /tmp/modules -o initramfs-sluinit-files.cpio ./cmds/* examples/sluinit
```

This decompresses the modules before adding them to the initramfs.

You might also need to re-write the dependency graph used by modprobe, but this should at least make insmod work.

rminnich, 2019-05-15:
adding a decompressor now

srcUp, 2019-05-15:
UEK Kernel has the following code that compresses the modules.
```
if [ "%{with_compression}" == "1" ]; then \
     find $RPM_BUILD_ROOT/lib/modules/ -type f -name '*.ko' -print0 | \
	xargs -0r -P 0 -n 1 /usr/bin/xz -f \
  fi \
```
Link to code here:
https://github.com/oracle/linux-uek/blob/v4.14.35-1916/uek-rpm/ol7/kernel-uek.spec#L1628

srcUp, 2019-05-15:
> Would it work if you build the initramfs like this instead?
I tried your suggestion above and executed the following commands in order
```
cp -RLp /lib/modules/`uname -r` /tmp/modules
unxz `find -name '*.ko.xz' /tmp/modules`
./u-root -format=cpio -build=source -files /tmp/modules -o initramfs-sluinit-files-uncomp.cpio ./cmds/* examples/sluinit
qemu-system-x86_64 -cpu host -kernel /boot/vmlinuz-4.14.35-1838.el7uek.x86_64 -initrd /home/Paul/go/src/u-root/initramfs-sluinit-files-uncomp.cpio -m 512 --enable-kvm -drive file=/home/Paul/Desktop/disk1,if=none,format=raw,discard=unmap,aio=native,cache=none,id=id1 -drive file=/home/Paul/Desktop/disk2,if=none,format=raw,discard=unmap,aio=native,cache=none,id=id2
```
I see the following screen at boot.
[image]

srcUp, 2019-05-17:
> I see the following screen at boot.
@rminnich pointed out that the above boot screen simply could be because of low memory given to QEMU. His suggestion and i quote
"almost always, it's an out of memory deal, if you are using qemu you might give it more memory"
worked great. Now I am able to boot into the u-root shell.

srcUp, 2019-05-22:
> But i don't see /tmp/modules directory inside u-root, so will look into it some more.
Since /tmp is mounted by u-root at init time, we cannot use /tmp/modules location for putting uncompressed modules.

srcUp, 2019-05-22:
Update: While building initramfs using u-root's binary, if i pass my host's modprobe binary via the `-files 'which modprobe'` parameter, modprobe works for me inside u-root shell.
However /bbin/modprobe fails. This means that the modprobe on my host machine(Oracle Linux 7.5) works fine but u-root's in house modprobe doesn't.

rjoleary, 2019-05-22:
There are some differences between u-root's modprobe and the others. In what way does u-root's modprobe fail? Is there an error message? There's an unsubmitted patch #1217 which adds xz compression support; where you using this?

Another thing to try is running the strace command on both the versions of modprobe to see where they differ.

Vanalytics, 2019-05-22:
@rjoleary the first comment when this issue was opened contains the error message modprobe emits. Yes, I am using the xz compression patch but modprobe still fails. Using the patch, insmod also fails because of unresolved dependencies. I had to plumb the host’s depmod (on rminnich’s suggestion) and run depmod -a to make insmod work. 

Strace hasn’t been helpful since modprobe is unable to find module path. So no system calls are made after that error is encountered.
Here is the strace with debugging enabled if you want to take a sneak peak
https://github.com/srcUp/OneTimeMisc/blob/master/strace-d-modprobe

srcUp, 2019-05-22:
@rjoleary the first comment when this issue was opened contains the error message modprobe emits. Yes, I am using the xz compression patch but modprobe still fails. Using the patch, insmod also fails because of unresolved dependencies. I had to plumb the host’s depmod (on rminnich’s suggestion) and run depmod -a to make insmod work. 

Strace hasn’t been helpful since modprobe is unable to find module path. So no system calls are made after that error is encountered.
Here is the strace with debugging enabled if you want to take a sneak peak
https://github.com/srcUp/OneTimeMisc/blob/master/strace-d-modprobe

rminnich, 2019-05-24:
actually it was not an ENOENT issue. There were lots of undefined symbols and, linux being an errno kind of kernel, the only error it had to return was ENOENT. So "no such file or directory" can be many, many things. Thanks errno :-)

rjoleary, 2019-06-26:
To whoever works on this bug next, @rminnich was working on a PR which just needs testing and rebasing: https://github.com/u-root/u-root/pull/1217

commit 7b2ea3ff (2020-09-12): kmodule: Add gzip and xz compressed kernel module support

Support gzip and xz compressed kernel modules in kmodule and thus
in modprobe and insmod.

Adds github.com/ulikunitz/xz dependency and reuses existing dependency
github.com/klauspost/pgzip.

Implemented in kmodule.FileInit instead of kmodule.LoadModule like #1217.

Uses kmodule.Init (eg. unix.InitModule) instead unix.FinitModule when
module is compressed. Thus the contents of the compressed module are read
into memory and passed as a slice of bytes to unix.InitModule. No messing
with file handles required as would be needed unix.FinitM [... cut]
closed 2020-09-12 by hugelgupf