u-root/u-root #1261: cpio fails when a file is inaccessible

rminnich, 2019-06-26:
rminnich@xcpu:~/projects/oreboot/oreboot/src/mainboard/ast/ast25x0$ find /sys/bus/pci | cpio o > /tmp/x
2019/06/26 01:05:46 Writing record "/sys/bus/pci/uevent" failed: open /sys/bus/pci/uevent: permission denied

our cpio quits at that point.

ls -l /sys/bus/pci/uevent
--w------- 1 root root 4096 Jun 24 16:08 /sys/bus/pci/uevent

 Would be nicer if it skipped over them. One does not always have the luxury of removing write-only files in a tree but would still wish to archive the rest of tree.

hugelgupf, 2019-06-26:
I don't see why you would want sysfs synthetic files in a cpio. This is not a bug to me, unless you can find a case that doesn't involve synthetic files.

With your patch, our cpio would write 4k of zeroes for a synthetic file what will be mounted over anyway. What's the use case for that?

If you want this feature, it should be a flag to cpio. "--include-synthetic" or "--include-inaccessible"

rminnich, 2019-06-27:
this has nothing to do with file size. Consider this:
 2219  touch /tmp/x
 2220  chmod 0200 /tmp/x
 2221  mkdir /tmp/y
 2222  rsync /tmp/x /tmp/y
rsync /tmp/x /tmp/y
rsync: send_files failed to open "/tmp/x": Permission denied (13)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1196) [sender=3.1.2]

note text: some files were not transferred. rsync, tar, and cpio do the best they can. Our cpio fails very badly.

hugelgupf, 2019-06-27:
> note text: some files were not transferred. rsync, tar, and cpio do the best they can. Our cpio fails very badly

Continuing with a log after failure, and including the file despite failure are two different feature requests. Which one do you intend here? 

Continuing after log makes sense to me.
Including file despite failure does not.

hugelgupf, 2019-06-27:
> note text: some files were not transferred. rsync, tar, and cpio do the best they can. Our cpio fails very badly.

Also, I always thought we do not intend to copy GNU commands bug-for-bug. I understand that what you're asking for is something they do.

What I've been asking you is *why* you want this. No write-only file in the world makes sense to include in a cpio, to me. Synthetic files do not make sense in a cpio.

Please answer the *why*, and "because the other ones do it" is not an answer.

hugelgupf, 2019-06-27:
Or let me rephrase that question. *What* are you using this for?

closed 2019-07-22 by rminnich