u-root/u-root #1262: cpio creates corrupt cpio files when a files reported size and read size differ

rminnich, 2019-06-26:
find /sys/bus/pci/slots/|grep address|wc
     64      64    1952

cpio alone in this case is u-root cpio.

find /sys/bus/pci/slots/|grep address | cpio o > /tmp/x

/bin/cpio -ivt < /tmp/x
-r--r--r--   1 root     root         4096 Jun 22 23:11 sys/bus/pci/slots/230/address
/bin/cpio: warning: skipped 20 bytes of junk
-r--r--r--   1 root     root         4096 Jun 22 23:01 sys/bus/pci/slots/38/address
/bin/cpio: warning: skipped 20 bytes of junk
-r--r--r--   1 root     root         4096 Jun 22 23:01 sys/bus/pci/slots/196/address
/bin/cpio: premature end of file

newc format may require padding to the reported size of the file, and we don't do that, so we miss most of the files in this archive.

hugelgupf, 2019-06-26:
Sorry, fat fingered the close button.

This only happens for synthetic files, which do not make sense in a cpio.

We should just error out on these, and/or provide a flag for the whacko behavior you want.

rminnich, 2019-06-27:
this can happen in many cases. The stat followed by a read is a race condition. Files can change size between the stat and the read, or even as you are reading them. The response of our cpio command is to create a corrupt file. rsync, cpio, tar, and in fact every file archiver I've ever used will handle it gracefully.

rminnich, 2019-07-22:
fixed.

closed 2019-07-22 by rminnich