u-root/u-root #1263: uio package lazy reader reports a close error even when it does not report an open error

rminnich, 2019-06-26:
The uio package lazy reader function can get an open error, should the program hand it a pathname of a file that can't be opened. It won't report it because it's a lazy open; it will be seen by the app as a read error, since the open occurs on first read. When the app tries to close the file, it can get a close error -- because the file in this case is not open. But the app did not get an open error. Getting a close error on a readonly file, when the open did not fail, is confusing.

The uio package lazy read function should not return close errors; or, at least, it could track when an open failed and in that case it can ignore the close error.

hugelgupf, 2019-06-26:
If the open() returns an error, lr.r should be nil.

What you're describing is *only* *ever* possible if

```go
func open() {
  return foostruct{}, fmt.Errorf("...")
}
```

Where have you seen such an open function

rminnich, 2019-06-27:
seen in the wild in our cpio command. It's in my closed pr.

hugelgupf, 2019-06-27:
Do you have a log or anything? Reading the code, I cannot imagine this is possible.