u-root/u-root #1276: Artifacts are sometimes not written

rjoleary, 2019-07-03:
The artifacts weren't written for this integration:

* https://circleci.com/gh/u-root/u-root/29444#artifacts/containers/0

The artifacts include the qemu serial logs, which makes this difficult to debug.

The error message is:

```
Uploading /tmp/serial to tmp/serial
  No artifact files found at /tmp/serial
```

Compare this to this integration test where the artifacts were written successfully:

* https://circleci.com/gh/u-root/u-root/29434#artifacts/containers/0

bjakobson, 2019-07-26:
Can it not find the artifact files within a folder or code, or is there an issue compiling it?

rjoleary, 2021-09-07:
We just write the whole serial output to the logs now. No separate files.

closed 2021-09-07 by rjoleary