u-root/u-root #1288: Flaky TestPxeboot

rjoleary, 2019-07-14:
This failure occurred on the master branch a few days ago: https://circleci.com/gh/u-root/u-root/29941

Relevant part of the log:

    integration.go:144: client: 2019/07/11 17:55:05 Bringing up interface eth0...
    integration.go:144: client: 2019/07/11 17:55:05 Attempting to get DHCPv4 lease on eth0
    integration.go:144: client: 2019/07/11 17:55:09 Attempting to get DHCPv6 lease on eth0
    dhclient_test.go:135: expect: timer expired after 30 seconds
    integration.go:144: client: [dhcpv4]2019/07/11 17:55:32 sent message: DHCPv4(xid=0x4ed82b68 hwaddr=0e:00:00:00:00:01 msg_type=DISCOVER, your_ip=0.0.0.0, server_ip=0.0.0.0)
    integration.go:144: server: 2019/07/11 17:55:32 Handling request from 0.0.0.0:68
    integration.go:144: server: 2019/07/11 17:55:32 Handling request DHCPv4(xid=0x4ed82b68 hwaddr=0e:00:00:00:00:01 msg_type=DISCOVER, your_ip=0.0.0.0, server_ip=0.0.0.0)
    integration.go:144: client: [dhcpv4]2019/07/11 17:55:32 received message: DHCPv4(xid=0x4ed82b68 hwaddr=0e:00:00:00:00:01 msg_type=OFFER, your_ip=192.168.0.2, server_ip=192.168.0.1)
    integration.go:144: client: [dhcpv4]2019/07/11 17:55:32 sent message: DHCPv4(xid=0x4ed82b68 hwaddr=0e:00:00:00:00:01 msg_type=REQUEST, your_ip=0.0.0.0, server_ip=192.168.0.1)
    integration.go:144: server: 2019/07/11 17:55:32 Handling request from 0.0.0.0:68
    integration.go:144: server: 2019/07/11 17:55:32 Handling request DHCPv4(xid=0x4ed82b68 hwaddr=0e:00:00:00:00:01 msg_type=REQUEST, your_ip=0.0.0.0, server_ip=192.168.0.1)
    integration.go:144: client: [dhcpv4]2019/07/11 17:55:32 received message: DHCPv4(xid=0x4ed82b68 hwaddr=0e:00:00:00:00:01 msg_type=ACK, your_ip=192.168.0.2, server_ip=192.168.0.1)
    integration.go:144: client: 2019/07/11 17:55:32 Got DHCPv4 lease on eth0: DHCPv4 Message
    integration.go:144: client:   opcode: BootReply
    integration.go:144: client:   hwtype: Ethernet
    integration.go:144: client:   hopcount: 0
    integration.go:144: client:   transaction ID: 0x4ed82b68
    integration.go:144: client:   num seconds: 0
    integration.go:144: client:   flags: Broadcast (0x8000)
    integration.go:144: client:   client IP: 0.0.0.0
    integration.go:144: client:   your IP: 192.168.0.2
    integration.go:144: client:   server IP: 192.168.0.1
    integration.go:144: client:   gateway IP: 0.0.0.0
    integration.go:144: client:   client MAC: 0e:00:00:00:00:01
    integration.go:144: client:   server hostname: 
    integration.go:144: client:   bootfile name: pxelinux.0
    integration.go:144: client:   options:
    integration.go:144: client:     Subnet Mask: ffffff00
    integration.go:144: client:     Router: 192.168.0.1
    integration.go:144: client:     DHCP Message Type: ACK
    integration.go:144: client: 2019/07/11 17:55:32 Boot URI: tftp://192.168.0.1/pxelinux.0

This issue looks different from #1209 and #1071

rjoleary, 2019-07-14:
This was right after #1282 was merged, but I can't see how that change has any effect.

hugelgupf, 2019-07-14:
I think the timer is just too short with one of my latest changes. I'll do something about it tomorrow

closed 2024-02-20 by hugelgupf