u-root/u-root #1312: Use golangci-lint instead of gometalinter in CI

rjoleary, 2019-07-29:
gometalinter is now deprecated. We use it here for running golint: https://github.com/u-root/u-root/blob/master/.circleci/config.yml#L91

See: 

* https://github.com/alecthomas/gometalinter
* https://github.com/golangci/golangci-lint

bjakobson, 2019-07-30:
Syntax wise they seem to be very similar if not the same. Key difference is how you call it..

golangci-lint = golangci-lint run

gometalinter = gometalinter // what ever you are calling, vendor, etc

I may be wrong, but I think other than calling each of them differently, the syntax is the same, so 
you should be able to just keep  the code and replace how you call it. Again, could be wrong on that.

bjakobson, 2019-07-30:
I was wrong, but right. The start is different, but the commands are almost the same.

golangci-lint = golangci-lint run --no-config --issues-exit-code=0 --deadline=30m 

gometalinter = gometalinter --deadline=30m --vendor --cyclo-over=30 --dupl-threshold=150

Other than that, the rest is the same. These are the commands that are interchangeable between the two: 

 --disable-all --enable=deadcode  --enable=gocyclo --enable=golint --enable=varcheck \
  --enable=structcheck --enable=maligned --enable=errcheck --enable=dupl --enable=ineffassign \
  --enable=interfacer --enable=unconvert --enable=goconst --enable=gosec --enable=megacheck

closed 2024-02-25 by hugelgupf