u-root/u-root #1327: integration: wait for VMs to be shutdown and logs drained before returning from test

hugelgupf, 2019-08-06:
```
panic: Log in goroutine after TestIntegration has completed

goroutine 505 [running]:
testing.(*common).logDepth(0xc000130300, 0xc000724030, 0x28, 0x3)
	/usr/local/go/src/testing/testing.go:634 +0x3a6
testing.(*common).log(...)
	/usr/local/go/src/testing/testing.go:614
testing.(*common).Logf(0xc000130300, 0x8382c9, 0x6, 0xc00000c040, 0x2, 0x2)
	/usr/local/go/src/testing/testing.go:649 +0x7f
github.com/u-root/u-root/pkg/vmtest.(*testLineWriter).OneLine(0xc0000e5fa0, 0xc000486041, 0x20, 0x3f)
	/go/src/github.com/u-root/u-root/pkg/vmtest/integration.go:150 +0x189
github.com/u-root/u-root/pkg/vmtest.jsonStripper.OneLine(...)
	/go/src/github.com/u-root/u-root/pkg/vmtest/integration.go:136
github.com/u-root/u-root/pkg/uio.(*fullLineWriter).printBuf(0xc000179c80)
	/go/src/github.com/u-root/u-root/pkg/uio/linewriter.go:33 +0x125
github.com/u-root/u-root/pkg/uio.(*fullLineWriter).Write(0xc000179c80, 0xc000b46000, 0x2, 0x2000, 0x2, 0x0, 0x0)
	/go/src/github.com/u-root/u-root/pkg/uio/linewriter.go:47 +0x20d
io.(*multiWriter).Write(0xc000204000, 0xc000b46000, 0x2, 0x2000, 0xc000b46000, 0x0, 0x0)
	/usr/local/go/src/io/multi.go:60 +0x87
github.com/u-root/u-root/vendor/github.com/google/goexpect.(*GExpect).read(0xc000b42340, 0xc00002a8a0, 0xc0005c6ff0)
	/go/src/github.com/u-root/u-root/vendor/github.com/google/goexpect/expect.go:1237 +0x132
created by github.com/u-root/u-root/vendor/github.com/google/goexpect.(*GExpect).goIO
	/go/src/github.com/u-root/u-root/vendor/github.com/google/goexpect/expect.go:1191 +0xa6
```

hugelgupf, 2019-09-14:
https://circleci.com/gh/u-root/u-root/32239

```
panic: Log in goroutine after TestIntegration has completed

goroutine 137 [running]:
testing.(*common).logDepth(0xc000102300, 0xc00002a4e0, 0x28, 0x3)
	/usr/local/go/src/testing/testing.go:634 +0x3a6
testing.(*common).log(...)
	/usr/local/go/src/testing/testing.go:614
testing.(*common).Logf(0xc000102300, 0x83bd28, 0x6, 0xc000764020, 0x2, 0x2)
	/usr/local/go/src/testing/testing.go:649 +0x7f
github.com/u-root/u-root/pkg/vmtest.(*testLineWriter).OneLine(0xc00000dfc0, 0xc0000243c1, 0x20, 0x3f)
	/go/src/github.com/u-root/u-root/pkg/vmtest/integration.go:150 +0x189
github.com/u-root/u-root/pkg/vmtest.jsonStripper.OneLine(...)
	/go/src/github.com/u-root/u-root/pkg/vmtest/integration.go:136
github.com/u-root/u-root/pkg/uio.(*fullLineWriter).printBuf(0xc000145c80)
	/go/src/github.com/u-root/u-root/pkg/uio/linewriter.go:33 +0x125
github.com/u-root/u-root/pkg/uio.(*fullLineWriter).Write(0xc000145c80, 0xc0007da000, 0x2, 0x2000, 0x2, 0x0, 0x0)
	/go/src/github.com/u-root/u-root/pkg/uio/linewriter.go:47 +0x20d
io.(*multiWriter).Write(0xc0001d6020, 0xc0007da000, 0x2, 0x2000, 0xc0007da000, 0x0, 0x1)
	/usr/local/go/src/io/multi.go:60 +0x87
github.com/u-root/u-root/vendor/github.com/google/goexpect.(*GExpect).read(0xc0006e4680, 0xc0011522a0, 0xc001515c00)
	/go/src/github.com/u-root/u-root/vendor/github.com/google/goexpect/expect.go:1237 +0x132
created by github.com/u-root/u-root/vendor/github.com/google/goexpect.(*GExpect).goIO
	/go/src/github.com/u-root/u-root/vendor/github.com/google/goexpect/expect.go:1191 +0xa6
FAIL	github.com/u-root/u-root/pkg/loop	18.598s
```

JulienVdG, 2019-10-23:
I'm not 100% sure but I think I solved this in tastevin by waiting on the channel from expect.SpawnWithArgs.
I just pushed a similar change to https://github.com/u-root/u-root/pull/1408 as this PR was hit by a similar error... let's see if it fixes it

JulienVdG, 2019-10-23:
Anyway I guess the bug is random, so a working run will probably not mean much.

JulienVdG, 2019-10-23:
Feel free to cherry-pick the commit if the rest of the (unrelated) PR needs more review

commit b6219edc (2019-10-23): pkg/qemu: Wait for goexpect end on close should fix #1327
closed 2019-10-23 by hugelgupf