u-root/u-root #1329: Preset vi .profile

bjakobson, 2019-08-07:
Much like making sure the go version is 1.12+, we need to make sure the GOROOT and GOPATH are hard coded to the vi .profile (or vim) to make sure that every time the computer runs, the gopath and goroot are set. Does this sound do-able?

should look something like this withing vi .profile:

# set PATH so it includes user's private bin if it exists
if [ -d "$HOME/bin" ] ; then
    PATH="$HOME/bin:$PATH"
fi

# set PATH so it includes user's private bin if it exists
if [ -d "$HOME/.local/bin" ] ; then
    PATH="$HOME/.local/bin:$PATH"
fi

export GOPATH=~/go
export PATH=$PATH:/usr/local/go/bin:$GOPATH/bin

Signed-off by: Brandon Jakobson <brandonjakobson@gmail.com>

hugelgupf, 2019-08-07:
I feel like this is more of a Go issue in general than a u-root issue. In fact, Go now sets a default $GOPATH, so you don't necessarily have to do the above anymore afaik

bjakobson, 2019-08-07:
Yes it is definitely more about Go than u-root. However, without this properly set up, you can only use u-root while the computer is on and the $GOPATH is set. Everytime you restart your computer, you have to reset the goroot and gopath in order to do anything that starts with command 'go'. I do not know about Go setting a default $GOPATH, because I could not do anything without setting mine, maybe it is a very new update.

msingle, 2019-08-07:
In Go's documentation related to $GOPATH (
https://golang.org/cmd/go/#hdr-GOPATH_environment_variable) it shows the
default is $HOME/go.
It appears the default was set in Go 1.8 (
https://golang.org/doc/go1.8#gopath)

bjakobson, 2019-08-07:
Huh, I had go version 1.10 for the majority of the time I have been coding Go, and I had to preset all mine, nothing was set without me actually hard coding it. Also, does this mean it is set in vi .profile as well? Other wise the GOPATH is only active while the computer is on..

closed 2019-08-07 by bjakobson