u-root/u-root #1368: ip6 dhclient doesn't accept Advertise on RapidSolicit

rminnich, 2019-09-18:
This is observed with 7d5452. pkg/dhclient/dhclient.go:lease6 does this:
        log.Printf("Attempting to get DHCPv6 lease on %s", iface.Attrs().Name)
        p, err := client.RapidSolicit(ctx, dhcpv6.WithNetboot)
        if err != nil {
                return nil, err
        }
RapidSolicit does this:
        msg, err := c.SendAndRead(ctx, c.serverAddr, solicit, IsMessageType(dhcpv6.MessageTypeReply))

So it will only take a Reply.

Here is what the ISC DHCP server does:

Solicit message from fe80::3e97:eff:fee9:ecd6 port 546, transaction ID 0xD7734200
Picking pool address 2001:db8:0:1::130
Advertise NA: address 2001:db8:0:1::130 to client with duid 00:01:00:01:25:15:6c:51:3c:97:0e:e9:ec:d6 iaid = 250211542 valid for 600 seconds
Sending Advertise to fe80::3e97:eff:fee9:ecd6 port 546

It sends an Advertise, which we reject, and we don't do a subsequent solicit, so the interface never comes up.

Parenthetically, the error handling is not helpful, it prints:
Could not configure eth0: no matching response packet received
followed by
Configured all interfaces

Well, no you didn't :-) 

Plus, "no matching response packet"? Some indication of what it wanted would be nice.

rminnich, 2019-09-18:
note: I'm working on a fix with Andrea for the packet issue but not for the error message issue (yet).

hugelgupf, 2020-03-18:
done with https://github.com/insomniacslk/dhcp/pull/356

closed 2020-03-18 by hugelgupf