u-root/u-root #1369: ip route show: show pretty routes

hugelgupf, 2019-09-20:
It's a bit of a pain to have to interpret these yourself:

```
ip: Iface       Destination     Gateway         Flags   RefCnt  Use     Metric  Mask            MTU     Window  IRTT                    
eth0    00000000        0100A8C0        0003    0       0       0       00000000        0       0       0
eth0    0000A8C0        00000000        0001    0       0       0       00FFFFFF        0       0       0            
eth0    0100A8C0        00000000        0005    0       0       0       FFFFFFFF        0       0       0
```

versus on real system:

```
default via 100.97.23.254 dev enp0s31f6 proto dhcp metric 100 
100.97.22.0/23 dev enp0s31f6 proto kernel scope link src 100.97.22.161 metric 100 
192.168.9.0/24 dev docker0 proto kernel scope link src 192.168.9.1 linkdown 
```

we can do something inbetween. at least display the IPs correctly, and decipher the flags.

10000TB, 2019-09-26:
seems like a change around fmt printing  ?

hugelgupf, 2019-09-26:
Pretty much, except we don't even parse the info in the first place:

https://github.com/u-root/u-root/blob/57a97d85efb63f79289bf6b9d102dd8559139bc3/cmds/core/ip/ip.go#L213

10000TB, 2019-09-26:
I'd be interested in taking this for fun ... seems a good start for newbie

/assign xuehaohu  ?

insomniacslk, 2019-09-29:
we have some working code for this, will ask my colleague to clean it up. IIRC Marcello also had some code for this, I'll ping him on Slack

insomniacslk, 2019-09-29:
as an additional note, we are moving away from vishvananda/netlink in favour of jsimonetti/rtnetlink/rtnl (safer, smaller)

sylvblck, 2019-09-29:
yeah i already started to work on it. #1354
But as @insomniacslk mentioned, we want to move to jsimonetti/rtnetlink/rtnl.
Right not it does not support all necessary attributes. So we need to work on it first.

There will be a WIP PR soon with the current progress :+1:

hugelgupf, 2019-09-29:
Make it two PRs. Showing the route has nothing to do with changing the netlink lib.

insomniacslk, 2019-09-29:
I need this for a demo next week, and my colleague's work is only for IPv6 routes, so I went ahead and created #1379 . However this is based on /proc/net/route like the original code, which does not provide as much information as rtnetlink. A better PR can come later

insomniacslk, 2019-09-29:
CC @sylvblck

insomniacslk, 2019-09-29:
I didn't reload the browser and didn't see @hugelgupf 's and @sylvblck 's replies until now. How about #1379 as a stop-gap until @sylvblck 's change is completed?

insomniacslk, 2019-09-29:
a quick chat with @sylvblck and looked at https://github.com/u-root/u-root/pull/1354 and my PR doesn't make sense anymore :) I'll close this in favour of reopening #1354

sylvblck, 2019-09-29:
I rebased and reopened #1354 :+1: it's working fine on my test setup

10000TB, 2019-10-01:
@sylvblck  mind sharing some local test setup ? - I can pick the ipv6 implementation since it related to my current work

hugelgupf, 2019-10-01:
I think the local test setup is... run it on your friendly local Linux machine (e.g. run it on your workstation and compare to system's ip route show...)

for the moment.

10000TB, 2019-10-01:
ah.. ok!  - ideally we like them be the same right ? at least the values if not the printing format

sylvblck, 2019-10-01:
@10000TB I'm using libvirt with `direct kernel boot` for each vm.
In this way it is possible to define host and network per xml file with virt-manager or by hand (for the specific use-case) and interact by using `virtsh`. Still not done with the setup but it may be worth for a blog post :smile:

10000TB, 2019-10-02:
i ill be interested in checkout out the post to come

not familiar with the libvirt - but sounds like you are starting vms per configs, and run ip cmd within vms to verify if they match what is in configs

if setup is checked in repo, it would be super helpful

10000TB, 2019-10-02:
Trying to find inspiration for ipv6 implementation, so far I found this https://github.com/shemminger/iproute2/blob/master/ip/iproute.c#L1691 - still reading and digesting

but meanwhile want to post it here and see if more experienced people have any other suggestion

hugelgupf, 2019-10-03:
Actually, I think @sylvblck's patchset is probably better inspiration for v6 than the C code: you should be able to call `netlink.RouteList` with family `FAMILY_V6`. Probably an easier thread to pull on than that C code.

sylvblck, 2019-10-03:
btw @10000TB  I wrote basic support to show IPv6 routes on my PR #1354, but there are still a lot of work to do. ( e.g. a better approach to format a route entry, add IPv6 route preference -> `pref`, entries where you add `mtu` ...)

10000TB, 2019-10-18:
noted

10000TB, 2020-02-14:
being dead on this for so long, trying to recollect and pick up original promises as I am coming back to work on network boot stuff ...

Fair to close this issue ? per  https://github.com/u-root/u-root/pull/1354

closed 2024-03-04 by hugelgupf