u-root/u-root #1370: ip route: support deleting routes

hugelgupf, 2019-09-20:
Support for both

- deleting individual routes
- clearing all routes

is really needed.

10000TB, 2019-10-02:
does this correspond to iproute flush  ? would this https://github.com/shemminger/iproute2/blob/master/ip/iproute.c#L1691 be a good reference for implementation

hugelgupf, 2019-10-03:
Yeah, that looks about right. I would choose to perhaps only implement the "table", "dev", and "protocol" filter mechanisms there. (also, just not having any filters at all!)

2SpaceMasterRace, 2023-10-02:
Hi @hugelgupf, this looks like something I could work on

is it cool if I get assigned to this so I can get started working

hugelgupf, 2023-10-02:
I think this might already be supported, we just forgot to close the issue

2SpaceMasterRace, 2023-10-02:
Ah too bad thanks anyways

hugelgupf, 2024-03-04:
Fixed by #1915