u-root/u-root #1470: Merge changes to cmds/exp without approval

rjoleary, 2019-12-03:
This came up on the mailing list the other week. How would everyone feel about implementing the following rule? How might it be implemented?

Pull requests to cmds/exp can be merge without approval if the following conditions are met:

1. Entire pull request is in cmds/exp
2. Passes all tests and code quality checks.
3. Has >80% test coverage with either new or changed code.
4. 7 days have passed without comment or code changes.

rminnich, 2019-12-03:
I think for now it could be done by hand. I.e. on a daily basis we look for PRs that fit this profile and just push them. If we ever get to higher volume we can look at automating int.

hugelgupf, 2019-12-03:
> Has >80% test coverage with either new or changed code.

I haven't seen any PR in exp that fits this pattern. Are we saying we're gonna step up our VM testing game?

rjoleary, 2019-12-06:
I agree the 80% code coverage is too ideal for exp where unpolished code is acceptable.

Would it make sense to drop #3 and instead have this rule?

* A respectable number of unit tests. As a reference, see cmds/exp/pox

I know it can be abused, but I'm hoping people will act in good faith.

rjoleary, 2021-06-07:
Particularly, this was for the cpu command -- it was moved to a different repository.

closed 2021-06-07 by rjoleary