u-root/u-root #1477: Enable dead code elimination to reduce binary size

rojer, 2019-12-05:
Go's dead code elimination algorithm is described and implemented [here](https://golang.org/src/cmd/link/internal/ld/deadcode.go).
TLDR is: elimination of unused public methods at link time will be inhibited if reflection function calling are used anywhere.

quick and dirty experiment shows that if DCE remains enabled, the uncompressed CPIO image size of just core goes down by ~2.3M: 15165376 -> 12707776, a 16% reduction.
with all exp commands, it's about the same: 19130524 - 16562332, about 14%

current uses of reflection in u-root are:
 1. elvish, to invoke built-in functions ([here](https://github.com/u-root/u-root/blob/master/cmds/core/elvish/eval/builtin_fn.go#L183)).
 2. dmidecode, to parse structs (addressed by #1476)
 3. text/template - uses reflection in struct attribute access, used in go/doc package, which is imported by go/build package.

(3) is the toughest to address, it's all in go core libraries.
but the payoff is significant, might be worth exploring.

rojer, 2019-12-05:
compressed (XZ) image size reduction: 4325624 -> 3684448 (15%)

rminnich, 2019-12-05:
I am confused, how do you invoke it? My familiarity with DCE was when Go's linker was Ken's linker from Plan 9, which just always did DCE. When did it become an option and why, I wonder? We should always do DCE; I just assumed we did.

rojer, 2019-12-05:
Go always does DCE, hen it can be sure that code is indeed dead.
use of reflection makes that impossible to say, as any code that seems dead may not really be, leading to runtime panics. so Go just disables DCE when reflection is used.
it used to be that use of anything from the reflect package would disable DCE, but over time that has been refined to just function/method invocation - see detailed description in the link above.
as described, i think it should be possible to get rid of these uses of reflection, re-enabling GCE and reducing the resulting binary size.

rojer, 2019-12-05:
https://pastebin.com/wYLyp58R is the quick and dirty hack for go std library and elvish that re-enabled DCE and that i used to get the numbers above.

rminnich, 2019-12-05:
so, wow, that is very interesting, I had no idea. Would it be worth extending the bb tool to warn when it sees code practices that disable gce? Or is there a tool usage that can do that for us?

rminnich, 2019-12-05:
Neat, if you want to put in that elvish fix I'll approve :-)

hugelgupf, 2019-12-05:
How does the fmt package affect us? Doesn't fmt use reflection, and fmt is used by nearly everything?

rojer, 2019-12-05:
@rminnich i think it would, but first we need to figure out what to do about the go std library deps.

re: elvish  - i'm pretty sure it breaks stuff, so no, it's not ready to be checked as is :) i'm not familiar with elvish enough (= at all) to be able to tell what exactly breaks. but since it's only builtin functions we're talking about, i guess they can be enumerated and the list made explicit, instead of just allowing anything via reflection?

@hugelgupf it does, but tno the parts that prevent DCE from working, i.e. not Value.Call or struct method lookup. it does seem that all that keeps us from getting DCE working is the text/template stuff, which i'm pretty sure is not too important.

however, since changes are all in the go runtime, i'm not sure what would be the best way to go about it. provide a patched runtime? this is not very user-friendly. but i guess could be done, e.g. provide a docker container with patched go that can be used to build smaller binary.
also, should it be raised as an issue upstream? maybe go can provide a build tag that disables method/function invocation in the template library, so users that care about binary size and are ok with not having that functionality (i.e. us) can build with that tag?

so, i can start by looking at elvish and coming up with a proper fix.
then, try to make a nice patch for go stdlib to build template w/o the naughty reflection bits and see if we can get any traction with it.

rjoleary, 2019-12-06:
Thanks @rojer! That's a really cool find.

For text/template, I don't grasp the dependency graph. I thought go/build was only used at build time?

Also, is there an easy test to determine if a binary was built without DCE? It would be helpful for the u-root build to print a warning because this could easily regress again in the future.

rojer, 2019-12-06:
@rjoleary so, i investigated the text/template use in u-root. turns out, the situation with unused code elimination in Go is pretty bad - unused global variables like maps and lists just cannot be eliminated (https://github.com/golang/go/issues/31704).
text/template happens to have a very expensive such map: a map of builtin functions, including `call`, which invokes reflect.Value.Call and thus disabled DCE.
it, and all the functions it references, are thus considered alive (which also retains a good amount of code), and DCE is disabled.

the template functionality is not even used anywhere: the way we end up with this anchor on our neck is through `cmds/core/installcommand`, which uses `pkgs/golang`, which uses `go/build`, which [invokes `go/doc.Synopsis`](https://github.com/golang/go/blob/69614c0d0e05787c8203bdc364c3293e1cf5094a/src/go/build/build.go#L889), which is a [short helper function](https://github.com/golang/go/blob/69614c0d0e05787c8203bdc364c3293e1cf5094a/src/go/doc/synopsis.go#L68) that does not use `text/template` at all, but other stuff in `go/doc` does and, because of the aforementioned bug, `text/template.init` gets retained, which retains `builtins`, which retains `call`, which uses `reflect.Value.Call`, which disabled DCE. ugh.

i filed https://github.com/golang/go/issues/36021 - that way, if `text/template.builtins` is not used, it is never initialized. this will solve this particular case.
i also commented on the bigger issue here - https://github.com/golang/go/issues/2559#issuecomment-562657812
i bet there are quite a few global vars that retain dead code, so if that is iimplemented, it's bound to shave off some more bytes.

as for ways to detect if DCE is working, it is possible to have a sentinel method on some struct that should normally be eliminated and have a check at the end of build if it has been - `go tool nm bb | grep DeadMethod` or some such.

hugelgupf, 2019-12-06:
With #1358 I can probably get rid of the go/build dependency on pkg/golang.

xaionaro, 2019-12-06:
Let me put my useless cents here :)

> text/template happens to have a very expensive such map: a map of builtin functions, including `call`, which invokes reflect.Value.Call and thus disabled DCE.

In my very humble opinion all such stuff that prevents DCE (like `call`) should be disableable via a Go build tag (like `forcedce` or `noreflect`). A patch to Golang with `noreflect` is very easy to prepare, but I doubt the upstream will accept it :)

In gcc there's `LTO` which does a lot of stuff, including DCE. And a wild not confirmed guess is if somebody will use `gccgo` or `llvm` then *may be* somebody will be able to use:
* Provided DCE on their level.
* Something like `-Oz`.

Also just an useless thought-food:
```
cat > hello_world.go <<EOF
package main

import "fmt"

func main() {
	fmt.Println("Hello world!")
}
EOF
xaionaro@ubuntu:~/test$ go build -compiler gccgo -o gccgo.bin hello_world.go 
xaionaro@ubuntu:~/test$ go build -o gc.bin -ldflags="-w -s" hello_world.go 
xaionaro@ubuntu:~/test$ strip -s *.bin
xaionaro@ubuntu:~/test$ ls -l
total 1432
-rwxrwxr-x 1 xaionaro xaionaro 1434968 Dec  6 22:42 gc.bin
-rwxrwxr-x 1 xaionaro xaionaro   23464 Dec  6 22:42 gccgo.bin
-rw-rw-r-- 1 xaionaro xaionaro      73 Dec  6 22:09 hello_world.go
```
```
xaionaro@ubuntu:~/test$ go build -o gc-debug.bin hello_world.go 
xaionaro@ubuntu:~/test$ go build -compiler gccgo -o gccgo-debug.bin hello_world.go 
xaionaro@ubuntu:~/test$ nm gc-debug.bin | grep -c FieldByNameFunc
13
xaionaro@ubuntu:~/test$ nm gccgo-debug.bin | grep -c FieldByNameFunc
0
```
Also there's `tinygo`, but it has very cropped standard library and does not fit `u-root` requirements even close.

rojer, 2019-12-06:
@xaionaro, ooh, this is very interesting. so gcc-built binary trims a lot more fat. i wonder how/if reflection works in gcc-go built binaries. it may be that LTO/DCE (--gc-section?) is disabled in that case. definitely something to look into.

xaionaro, 2019-12-07:
Eh... I've just tried to compile `u-root/bb` with gccgo and:
```
xaionaro@ubuntu:~/go/src/github.com/u-root/u-root/bb$ go build -compiler gccgo -gccgoflags "-Os -flto"; strip -s bb; ls -lh bb
-rwxrwxr-x 1 xaionaro xaionaro 18M Dec  7 00:20 bb
```
I also was have to do a lot of dirty hacks to make it work, but the size is even worse than with simple `go build -ldflags '-w -s'` :(

xaionaro, 2019-12-10:
Created a ticket: https://github.com/golang/go/issues/36073

xaionaro, 2019-12-10:
An update:
Good news! It appears GCCGO allows to reduce binary of size `15884288` to `12190424`.

The line to compile:
```
go build -compiler gccgo -gccgoflags=all='-flto -Os -fdata-sections -ffunction-sections -Wl,--gc-sections'
```

What is required to do to use it:
* Update Go-runtime of `gccgo` to add `crypto/ed25519` (or apply another workaround).
* Update Go-runtime of `gccgo` to add `crypto/poly1305` (or apply another workaround),
* Update Go-runtime of `gccgo` to add `os.(*File).SyscallConn()`.
* Fix `gccgo` or `github.com/u-root/u-root/pkg/boot/multiboot/internal/trampoline` to compile this `trampoline`. At the moment the problem:
```
# github.com/u-root/u-root/pkg/boot/multiboot/internal/trampoline
../pkg/boot/multiboot/internal/trampoline/trampoline_linux_amd64.s:10:10: fatal error: textflag.h: No such file or directory
 #include "textflag.h"
          ^~~~~~~~~~~~
compilation terminated.
#
```

rojer, 2019-12-14:
https://github.com/u-root/u-root/pull/1486 removes use of reflection from elvish without (i hope) breaking any of the functionality.
confirmed that with this and the [text/template patch](https://go-review.googlesource.com/c/go/+/210284/) DCE starts working:
```
[rojer@nbd ~/go/src/github.com/u-root/u-root/bb master]$ GOROOT=/home/rojer/go/go_210284 GOARCH=arm GOARM=5 go build -v && ll bb
-rwxr-xr-x 1 rojer rojer 18057313 Dec 14 18:32 bb
[rojer@nbd ~/go/src/github.com/u-root/u-root/bb elvish_noreflect]$ GOROOT=/home/rojer/go/go GOARCH=arm GOARM=5 go build -v && ll bb
-rwxr-xr-x 1 rojer rojer 18193874 Dec 14 18:33 bb
[rojer@nbd ~/go/src/github.com/u-root/u-root/bb elvish_noreflect]$ GOROOT=/home/rojer/go/go_210284 GOARCH=arm GOARM=5 go build -v && ll bb
-rwxr-xr-x 1 rojer rojer 15209315 Dec 14 18:33 bb
```

unfortunately, the aforementioned patch will be integrated into Go 1.15 which will be rleeased in aug 2020, so we need to find some way of incorporating it into builds before that. i'll look into it.

xaionaro, 2019-12-15:
An update from my side:

1. I was stupid and did not notice that `gccgo` (in contrast to `gc`) builds binaries which dynamically links to `libgo`, `libgcc`, `libc` and other. So even if DCE for Golang-code works better, when I add `-static` it still weights more. And in those libs there's a lot of stuff that should be eliminated (like `atanf`) but it is not (so it seems DCE works badly for those libs). In this context I tried linker `gold` instead of `bfd`, but it does not work with `gccgo` and `-flto` :(. Found a bug report of the same linking problem of rust community and they just excluded `gold`. So I decided this is a long way (while still do-able, IMHO) and I should put this idea away for a while. So now I'm trying to find an easier way to reduce the size of `bb`.

2. Another approach is to port it to `386` (from `amd64`). Preliminary experiment showed that the size of the binary reduce from 16MiB to 14MiB. So now I'm trying to make it work with `GOARCH=386` and I will create another issue or/and PR for this.

rojer, 2019-12-17:
for DCE to work on libraries, they need to be built with -ffunction-sections as well, which they may not already be.
GOARCH=386 will definitely help a lot - instructions are more compact and it will reduce symbol table quite considerably (will be similar to arm, which is much smaller - see my quick experiment [here](https://groups.google.com/d/msg/golang-nuts/hEdGYnqokZc/HCPBlmqoAgAJ)).

rojer, 2019-12-20:
as an interim solution, i have created a docker image that contains go with patched stdlib - https://github.com/u-root/u-root/pull/1491

it's pretty easy to use and delivers the result:

```
[rojer@nbd ~/go/src/github.com/u-root/u-root/tools/golang_patched golang_patched]$ docker build -t golang-patched .
Sending build context to Docker daemon  9.728kB
Step 1/4 : FROM golang:1.13.5-alpine
...
Successfully built 794bb9635caf
Successfully tagged golang-patched:latest
[rojer@nbd ~/go/src/github.com/u-root/u-root/tools/golang_patched golang_patched]$ docker run --rm -it -u $(id -u):$(id -u) \
>       -v $(go env GOPATH)/src:/go/src \
>       -v $(go env GOCACHE):/go/.cache \
>       -v $PWD:/out \
>     golang-patched sh -c 'go build github.com/u-root/u-root && \
>                           ./u-root -build=bb -o /out/initramfs'
2019/12/20 14:14:54 Disabling CGO for u-root...
2019/12/20 14:14:54 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/go CGO_ENABLED=0
2019/12/20 14:14:54 Filename is /out/initramfs
2019/12/20 14:15:03 Successfully wrote initramfs.
[rojer@nbd ~/go/src/github.com/u-root/u-root/tools/golang_patched golang_patched]$ ll initramfs 
-rwxr-xr-x 1 rojer rojer 12666816 Dec 20 14:15 initramfs
```

xaionaro, 2020-01-14:
A partly related problem: https://github.com/golang/go/issues/36313

xaionaro, 2020-01-15:
Flag `-gcflags=all=-l`  disables function inlining and it allows to reduce the binary on 9.5% in the uncompressed state and 15% in a compressed state.
```
-rwxr-xr-x. 1 experiment0 experiment0 14417920 Jan 15 11:08 bb-noinline
-rwxr-xr-x. 1 experiment0 experiment0 15917056 Jan 15 11:08 bb
-rwxr-xr-x. 1 experiment0 experiment0 3835208 Jan 15 11:08 bb-noinline.xz
-rwxr-xr-x. 1 experiment0 experiment0 4508904 Jan 15 11:10 bb.xz
```

rojer, 2020-01-15:
wow, awesome!

> May be we should create a metatask for any size-reduction methods or rename this task?

yes, i think so.

xaionaro, 2020-01-16:
@rojer: JFYI, I summarized all found ways to reduce a Golang binary here:
https://github.com/xaionaro/documentation/blob/master/golang/reduce-binary-size.md

rojer, 2020-08-27:
so, DCE is working, we are using it and https://github.com/u-root/u-root/issues/1808 added a test to make sure we don't regress.

closed 2020-08-27 by rojer